open-ephys/GUI #232: 16 channel headstage(Intan RHD2132)

young-gyun, 2015-01-05:
I am using Intan`s acquisition board(RHD2000) with 16channel RHD2132 amplifier board(Intan). When I display signal in GUI with RHYTHM FPGA source tab set to 16channel,  only 8 of them display signal(CH 9-16). When I set it to 32, all 16channel signals are displayed(from CH9 to 24), however, I cannot know the wiring(which channels in GUI belongs to which physical channel in RHD2132(. Why is it not automatically recognized as 16channel headstage and what is the wiring in this case?

koreign, 2015-01-05:
Hey

Since the 16chan is based on the 32 chip, you should actually use the 32ch setting for it. Then as you mentioned there is a set of channel mapping to do it. I provided the configuration file for the 16ch mapping as well as the mapping in case you need 2x 16chan plugged in.

Please look for the configuration file Here:
https://github.com/open-ephys/GUI/tree/master/Resources/Configs

koreign, 2015-01-05:
Note there are empty channels because of:
1. The original mapping of a 32chan
2. The 3axis accelerometer is not present on 16ch

young-gyun, 2015-01-06:
Thank you for your reply! Just in case, Could I use the config file for
Intan2216 amplifier?

Thank you in advance!

Best,
Young

young-gyun, 2015-01-06:
I tried this but it looks not working in GUI ver 0.3.3(32bit)... do you
just in case have this file for that version?

jsiegle, 2022-07-30:
By clicking on the appropriate port icon in the Rhythm Plugin, it's now possible to toggle between the 16-channel and 32-channel headstages.

closed 2022-07-30 by jsiegle