open-ephys/GUI #235: GUI interfacing

nikolaskaralis, 2015-01-07:
What is the state of the GUI interfacing with Matlab/Python/Julia?

A search in the repository didn't reveal any modules that would allow this so far.

Embedding Python ([link] (https://docs.python.org/2/extending/embedding.html)) or Julia [link (http://julia.readthedocs.org/en/latest/manual/embedding/)] seems not too complicated.

For starters, we could have a "filter" module which receives the signals and allows you to load a .py or .jl file. The module would call then Python/Julia to perform the function specified in the script file on the data.

Seems less complicated initially than the Matlab Engine approach.

jsiegle, 2015-01-07:
Julia interfacing already works, check out the "jvoigts" branch (https://github.com/open-ephys/GUI/tree/jvoigts). Don't think it's well documented, though. @jvoigts, care to comment?

Python plugins, though slightly more complicated, also seem to be working in Francesco Battaglia's fork: https://github.com/fpbattaglia/GUI

Our plan for the next few months is to maintain a stable version of the GUI with basic functionality in the current master branch, and move most of the new development (other than bug fixes) to a version with a true plugin architecture. At that point, we'll make Python and Julia plugins supported features.

In the meantime, you can play around with these versions and see if they do what you need.

jvoigts, 2015-01-07:
Yea i have a working, very basic julia interface in that branch, it is missing some key features though and is pretty much undocumented. 
The key thing to do is to add event in/output - there are a few ways to do this, i believe the easiest on the user side would be to just have a binary samples x event channels array that the julia code can work on to implement any conditional processing/ event output.

I can add some documentation/ installation instructions asap if you want to give it a spin - the basics are all in the julia doc: http://julia.readthedocs.org/en/latest/manual/embedding/

nikolaskaralis, 2015-01-07:
That is great news. I will give it a try in the next few days.
Some basic documentation would be great when you have the time. :+1: 
Given that the branch is somewhat old (before the major update) do you think this implementation will break with the latest changes? 
For the events, indeed that sounds like a reasonable solution.

allenyin, 2016-01-30:
I came across the announcement about the GUI plugin demos (C/C++, Python, Julia) at SfN'15 (http://jvoigts.scripts.mit.edu/blog/open-ephys-poster-and-meeting-sfn2015/), unfortunately I was not at the conference to see it.

Where can I find more information about using the Julia plugin?

jvoigts, 2016-01-30:
Hi,
that code is on https://github.com/open-ephys/GUI/tree/jvoigts
this is currently in _very_ rough weekend-project form (it works for passing neural data back and forth, acting as a filter, but that's it), there's significant work needed to make this work with the current julia version, to enable event passing between julia and the GUI, and to generally clean things up etc. If you plan on doing any work on this, i'd recommend starting by re-implementing the existing code in a proper plugin format for the plugin GUI ( https://github.com/open-ephys/plugin-GUI ) as well.

Happy to answer questions etc.! It would be great to get this plugin on the road properly. @nikolaskaralis also has put time into this more recently and likely knows more about what is up with newer julia releases etc.

allenyin, 2016-01-31:
Ok, I will take a look at it.

A somewhat related question...how does the openephys GUI generally fit into closed-loop BMI experiments? Most of the documentation so far seems to only show its uses in spike-sorting and recording raw waveforms and spikes to disk.

In a closed-loop experiment setting, I would want to stream the raw waveforms and spikes to an experimental control system, which processes that and then output actions to either the acquisition board GPIO through the GUI, or to other actuators through custom hardware.

I don't think it will be too hard to write a plugin to allow streaming of the acquired data from the GUI out (or is it? I'm still at the research stage). If the external application is written in a higher level language than C++, say Julia, to leverage the ease of implementing mathematical algorithms, then I'm wondering how would the real-time processing speed compare to if the external program was also written in C++?

jsiegle, 2022-07-30:
In most cases, we recommend using the [ZMQ Interface](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/ZMQ-Interface.html) to stream data to other applications.

There's also the [Matlab Interface](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/MATLAB-Interface.html), which can stream a single channel to Matlab and then back to the GUI. 

To read a data stream out of Bonsai, you can use the [Ephys Socket](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Ephys-Socket.html) plugin.

If you have specific questions about inter-process communication, please open a new issue in the [`plugin-GUI`](https://github.com/open-ephys/plugin-GUI) repo.

closed 2022-07-30 by jsiegle