open-ephys/GUI #237: SpikeSorter issues

nikolaskaralis, 2015-01-08:
Building the latest version of the source, the SpikeSorter module crashes then GUI (problem with the Juce Audio and segmentation fault). With the source version of a few weeks back, things work fine.

Btw, In the spike sorter, the prev/next electrode buttons are not working.

aacuevas, 2015-01-08:
I can't reproduce this right now. What steps cause it to crash for you? What SO version?

Also, prev/next work for me, they cycle between the created electrodes.

nikolaskaralis, 2015-01-09:
Just putting a Spike Sorter after the Rhythm Module and adding 16/32 channels in the Spike Sorter is enough. I am on Ubuntu 14.04.1 LTS

When I click acquisition the error is:
Starting acquisition.
Flushing FIFO.
   Enabling VisualizerEditor
SpikeSorterCanvas beginning animation.
Filter Viewport disabled.
Error in Rhd2000EvalBoard::selectDacDataStream: stream out of range.
Error in Rhd2000EvalBoard::selectDacDataChannel: dataChannel out of range.
Error in Rhd2000EvalBoard::selectDacDataStream: stream out of range.
Error in Rhd2000EvalBoard::selectDacDataChannel: dataChannel out of range.

Adding audio callback.
JUCE Assertion failure in juce_AudioSampleBuffer.cpp:486
JUCE Assertion failure in juce_AudioSampleBuffer.cpp:488

open-ephys, 2015-01-09:
It was caused by a silly mistake in processing the overflow buffer. The indexing variable should have been "j" instead of "i".

Prev/next electrode buttons seem to be working for me.

closed 2015-01-09 by open-ephys

nikolaskaralis, 2015-01-09:
Great. The Prev/next electrode button was for the previous version of the code (since I couldn't test the new one due to this crash). As I see in another comment, this has been solved in the meantime.