open-ephys/GUI #241: Apply referencing only to data channels in CHANNEL MAP Module

young-gyun, 2015-01-15:
Hi! I found that referencing by select one channel in CHANNEL MAP module PARAMS tab applies also to AUX channels. is there any way to apply referencing only to all or some of data channels? I tried to utilise splitter and apply difference referencing by adding different CHANNEL MAP but it crashes. 

Just in case, I am using Intan RHD2000 board.

young-gyun, 2015-01-15:
In general, lots of problems occurs because AUX(ADC)channels are included with same data flow with 'DATA' channels(in my case, at 'A' of RHYTHM FPGA because I only use one head stage). How about discriminate their flow at initial point?

open-ephys, 2015-01-15:
You can control which channels are referenced by selecting or deselecting their numbers in the main segment of the Channel Map module. This way you can avoid referencing the AUX channels. Does that help?

It would be nice to figure out where the crashes are coming from, though. Can you upload the configuration file for your signal chain?

jsiegle, 2022-07-30:
The Channel Map plugin no longer allows referencing. For digital referencing, you should use the [Common Average Reference](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Common-Average-Reference.html) or [Virtual Reference](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Virtual-Reference.html) plugins.

closed 2022-07-30 by jsiegle