open-ephys/GUI #243: Unable to install v 0.3.4 on Mac OS 10.10 both binaries and source code compiler

DrTonus77, 2015-01-28:
I installed v 0.3.3 on my Mac last month with success. Everything worked fine. I tried to install the new GUI through the binary file, completed all the instructions and I got the following error message:
Dyld Error Message:
  Library not loaded: /opt/local/lib/libzmq.3.dylib
  Referenced from: /Applications/Open Ephys/open-ephys.app/Contents/MacOS/open-ephys
  Reason: image not found

I then moved on to try to run the GUI from source code. Installed Xcode, MacPorts, HDF5 librairies, etc...

Upon compiling in Xcode, besides the 23 Unused Entity issues, it crashes at the following step:
in NetworkEvents.h (marked by ***)

#ifndef __NETWORKEVENT_H_91811541__
#define __NETWORKEVENT_H_91811541__
//#define ZEROMQ

#ifdef ZEROMQ 
	
#ifdef WIN32
	//#pragma comment( lib, "../../Resources/windows-libs/ZeroMQ/lib_x64/libzmq-v120-mt-4_0_4.lib" )
	#include "../../Resources/windows-libs/ZeroMQ/include/zmq.h"
	#include "../../Resources/windows-libs/ZeroMQ/include/zmq_utils.h"
#else
    #include <zmq.h>     **********   zmq.h file not found *********

My search paths are correct and include the HDF5 librairies installed by MacPorts

Header search path:
HEADER_SEARCH_PATHS = ../../JuceLibraryCode ../../JuceLibraryCode/modules /opt/local/include /usr/local/include $(inherited)
//:completeSettings = none

Library search path:
LIBRARY_SEARCH_PATHS = $(inherited) "/opt/local/lib" "/usr/local/lib"

Any help would be appreciated,

Sébastien

open-ephys-GUI-binaries, 2015-01-28:
Hi Sébastien,

The GUI depends on the ZeroMQ library now. Once you install it via MacPorts (sudo port install zmq), the GUI should run and compile.

I'll add something about this in the documentation later today—this was a recent change, and we haven't updated it yet.

Josh

jsiegle, 2015-01-28:
Hi Sébastien,

The GUI depends on the ZeroMQ library now. Once you install it via MacPorts (sudo port install zmq), the GUI should run and compile.

I'll add something about this in the documentation later today—this was a recent change, and we haven't updated it yet.

Josh

DrTonus77, 2015-01-28:
Thanks Josh, that was a quick fix. So far so good.

closed 2015-01-28 by DrTonus77