open-ephys/GUI #244: New user: "file type not supported"

JMHyman, 2015-02-06:
I just downloaded the Windows version and I'm trying to walk through the tutorial, but when I try to open one of the data files in the File Reader it fails. At the bottom of the GUI it says "File type not supported". Has anyone else had this problem?

jsiegle, 2015-02-06:
Looks like we forgot to update the files for that repository. Try downloading it again, it should work now.

JMHyman, 2015-02-06:
So I just downloaded the Windows zip file again, deleted my old version and then the same thing happens (File type not supported).

jsiegle, 2015-02-06:
Try it now. I mistakenly only updated the files for the 32-bit version.

cristi-zz, 2018-11-21:
Hi! I have the same issue, ubuntu 18.04, fresh install of GUI. I started the tutorial but when I load ```open-ephys-linux/DataFiles/data_stream_16ch_cortex.kwd``` I get the ```File type not supported``` message in bottom-left side of the screen. What can I do?

aacuevas, 2018-11-21:
Can you run the GUI from the console and check the error messages?
Nonetheless, I believe that might be due to the hdf5 library.
The GUI works with version 1,8 of hdf5. When they updated it to 1.10 they broke compatibility in some areas, so it is far from trivial for ut so make the GUI work with both versions.

Ubuntu 18 ships with 1.10, whereas any other versions come with 1.8, so it is a known issue we have still pending.

cristi-zz, 2018-11-21:
Oh, yes, @aacuevas you are correct! I was about to post a wacky solution. 
Indeed, the culprit is HDF5 "driver" required by KWIKFormat and NWBFormat plugins. My wacky workaround:

 - Install HDF5 if it is not installed using ```sudo apt-get install libhdf5-serial-dev```
 - Run ```./open-ephys``` from console and check the error messages. It will complain (in turn) about two files.
 - Do a ```find``` for those files, removing the version digits. Find the "closest" match
 - Copy the file(s) in ```open-ephys-linux``` folder, renaming them as the missing files. One example: ```cp /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.100 libhdf5_serial.so.10```
 - Disable hdf5 version checker (eg you are using one version for the headers and another one for the static libraries) by ```export HDF5_DISABLE_VERSION_CHECK=1```
 - Run the open-ephys again and note that the plugins are loaded.

You got a ton of messages when you try to use the file reader plugin but I got some realistically looking data.

It would be possible that all these errors go away if one just compile locally the software.

Cheers,
Cristi

LE. It *might* be possible to install hdf5 using anaconda (this is a python env). In anaconda one can have fine control over what versions get installed.

closed 2022-07-30 by jsiegle