open-ephys/GUI #245: yosemite master not working

rjs360, 2015-02-10:
I installed the binary from the Open Ephys software page, copied the three .dylib files to the /usr/lib directory, but running the application file after this gives me an error on Mac OS 10.10.2;

"open-ephys cannot be opened because of a problem. 
Check with developer bla bla bla etc. "

Any thoughts on how to solve this?  Build from scratch?

jsiegle, 2015-02-10:
Maybe try copying those files to /opt/local/lib instead? I'm not sure what else the problem could be.

Can you paste the exact error message?

rjs360, 2015-02-10:
Hi Josh,

I built it from scratch using homebrew which, after installing hdf5 and 
the other file with macports, now works fine.

Then I tried installing the binary again, and it now works too...
must have been missing some files.

Ruud

> Josh Siegle <mailto:notifications@github.com>
> February 10, 2015 at 1:29 PM
>
> Maybe try copying those files to /opt/local/lib instead? I'm not sure 
> what else the problem could be.
>
> Can you paste the exact error message?

koreign, 2015-03-19:
I also had the same problem in installing the binary. I think that the wiki should mention about installing macports, updating and installing the QMzero thingy no ?

jsiegle, 2015-03-20:
Is MacPorts necessary, or can you get it to run by copying the .dylib files included in the downloaded directory?

closed 2022-07-30 by jsiegle