open-ephys/GUI #246: Impedance measurement crash

nikolaskaralis, 2015-02-12:
When measuring impedance manually, things work ok.
But when selecting the impedance measurement at acquisition start and you start acquisition the system crashes. Should be easy to reproduce (Ubuntu 14.04).

Also, it seems that in the last version of the GUI, the impedance measurement result is not displayed anywhere in the GUI. It is displayed however in the console. If I remember correctly it used to be displayed. I think it was a nice feature.

mikailweston, 2015-02-12:
I have a similar issue that impedance measurement manually or at acquisition start causes the GUI to crash every time.
I'm using the latest version of the GUI compiled on Windows 7 in VS2013 and using intan headstages.

aacuevas, 2015-02-12:
I'm not getting any crashes (on Windows 7), but it seems that the Rythm canvas has broken, that might be the cause... I'll let you know when it's fixed to see if it continues crashing then.

jsiegle, 2022-07-30:
The impedance measurement interface has been redesigned in the latest version of the GUI; it should be more reliable now.

closed 2022-07-30 by jsiegle