open-ephys/GUI #248: LFP viewer hide settings

nikolaskaralis, 2015-02-12:
When you hide some channels in the LFP viewer, if you double click on one channel to go to the single channel view and then you go back to all the channels view, the hidden channels reappear.  It would be great if this hidden flag is preserved throughout the session (and ideally saved in the template settings or in some settings xml file dedicated to the LFP viewer settings as well).

nikolaskaralis, 2015-02-12:
On a related topic: If I recall it used to be possible to disable channels completely.
In the last version, I am trying to completely disable channels (at both the FPGA and LFP viewer modules, by deselecting them from the params menu) but it seems I am not able to do it.
Is it still possible? If yes, how can it be done?

aacuevas, 2015-02-12:
I'd swear I already fixed this some time ago... I'll look into it again.

For completely disabling channels, you can use the channel mapper, just double click a channel in remap mode.

nikolaskaralis, 2015-02-12:
Thanks, the disabling works like that.
Speaking of channel numbers: I realized that when you disable/rearrange some channels, in the LFP viewer they propagate with new channel numbers (from 1 to total number of channels). However this can get very confusing for the non-trivial cases, since the number that you see in the LFP viewer has no correspondence to the actual channel number and is instead just the nominal channel order.

I believe it would be much clearer if the channel number that the LFP viewer receives/displays is the original channel number (similar to the way the channel number are rearranged in the Rec tab of the channel map module).

aacuevas, 2015-02-12:
Just recently I've got feedback on the opposite direction (channels being saved with the original number instead of the reordered one), so it seems that this is one of those things that depends on personal preference. I'm thinking on adding a rename/renumber toggle on the channel mapper (and perhaps the LFP viewer), but I have to figure out the best way to do it.

aacuevas, 2015-02-12:
Toggling single channel mode should not reactivate channels now.

closed 2022-07-30 by jsiegle