open-ephys/GUI #250: losing data

mikailweston, 2015-02-12:
I'm having a really annoying issue where variable numbers of channels are recorded. Using the intan rhd2032 the GUI records only channels 1 to 4. Channels 5 to 32 are zeroes. 

I have set the GUI to record only from the rhythm fpga.

aacuevas, 2015-02-12:
This is strange indeed. What system and GUI version are you using? Could you please attach a saved configuration which has this issue?

mikailweston, 2015-02-12:
Hi,

config7 records only 7 channels

config5 records only 5 channels

Thanks,

Mikail

mikailweston, 2015-02-12:
The other issue I get is the the GUI crashes every time I stop recording, but only when recording in the kwik file format. I don't know if this is a related problem.

aacuevas, 2015-02-12:
I can't find any attached files, please use the github page to attach them, it seems email attachments get filtered.

Also, which operating system are you using? Are you using a prebuilt binary or are you building the GUI yourself?

mikailweston, 2015-02-12:
Github only allows uploading of jpg, png or gif  file types.

I updated my comment on github with the configuration files by renaming them with the .jpg file extension. You can download the files above and view them in a text editor

I'm using Windows 7, and compiling the binary myself using VS2013 with the latest updates.

aacuevas, 2015-02-12:
config5 was written with 0.3.3, so I can't open with the current branch. However, I'm able to reproduce your issues (both the zero saving and the crash) with config7. Seems like the channel mapper is affecting saving when it shouldn't. I'll look into that.

aacuevas, 2015-02-14:
It should be fixed now. Please check if it works for you.

mikailweston, 2015-02-15:
Hi 

I am getting a lot of build errors now. 

Ihave installed the ecube api but am getting many errors in ecubethread.cpp when I try to build.

mikailweston, 2015-02-15:
I reinstalled the ecube api and it all works now. 
It doesn't look like i am losing data anymore
Thanks!

jsiegle, 2015-02-15:
Sorry about that...the #ECUBE_COMPILE flag should be disabled, that's for testing with another system.

closed 2015-02-18 by aacuevas