open-ephys/GUI #252: Settings xml saving

nikolaskaralis, 2015-02-17:
in many cases, the settings.xml file is not being saved. However I could not figure out yet what causes this behavior.

aacuevas, 2015-02-19:
Do you mean the auto saved settings or when manually saving a signal chain?

nikolaskaralis, 2015-02-19:
The auto-saved settings.xml file that is supposed to be contained in each
recording directory.

aacuevas, 2015-02-25:
I can't reproduce this, at least on Windows. If you find a way to provoke this issue please tell me.
I have, however, fixed a related issue: we were generating a settings file for only the first "experiment" (acquisition) per folder, even if the chain was changed. I've fixed that so, perhaps, that also fixes this. Please check it if you can.

nikolaskaralis, 2015-02-27:
I don't have a particular way to check it (not sure when exactly it was happening) however lately when recording the settings.xml appears normally so we can consider it fixed for the moment.

closed 2015-02-27 by nikolaskaralis