open-ephys/GUI #253: Impedance file names

nikolaskaralis, 2015-02-17:
In the impedance file, the following is being written:
<CHANNEL name=" &#6;M&#8;" stream="0" channel_number="0" magnitude="8815934"
           phase="-107.42929840087890625"/>

I ran into problems while trying to process this file, since this string in the name field is not conforming to xml standards. What is it supposed to be? Is it correct?

aacuevas, 2015-02-18:
The impedance files stores both magnitude and phase of the impedance as attributes for each channel. that's the result of a simple call to a Juce's function: chan->setAttribute("phase",phase[i])

Why is that string not compliant with xml standard?

nikolaskaralis, 2015-02-18:
I just noticed that github markdown didn't parse correctly what I had copy-pasted.
Here is the full string:

     <CHANNEL name=" &#6;M&#8;" stream="0" channel_number="0" magnitude="8815934" phase="-107.42929840087890625"/>

The problem is with the name field.

aacuevas, 2015-02-18:
Ok, that's definitely not normal, it should be the channel name. That's probably a segfault waiting to happen.
Recently a couple of "advanced features" of the Rhythm node broke, and this seems to be another of those. At the moment the rhd2164 branch has some changes to the processor code, so I'd prefer to wait until those are tested and merged back before working on those to avoid breaking even more stuff but as soon as it's merged I'll work on those.

jsiegle, 2022-07-30:
This has now been fixed.

closed 2022-07-30 by jsiegle