open-ephys/GUI #254: Ecube support

shayo, 2015-02-18:
FYI, the ECUBE_COMPILE is (still?) defined in VS2013 and causes compilation errors (specifically, on
#import "libid:60C0AAC2-1E0B-4FE5-A921-AF9CEEAAA582")

mborisov1, 2015-02-18:
Is this a problem? It's just that I always compile OpenEphys with ECube support and therefore I define that flag. The #import directive will not produce errors if you install EcubeAPI.

If you think it's a problem, I could deactivate the ECUBE_COMPILE flag before committing the .vcproj file to the repository.

shayo, 2015-02-18:
Just a thought. Most users out there don't have the Ecube, and it doesn't
say anywhere you need to install the API to compile the project.
I would recommend to either install the eCube API by default, or to set
that flag to zero by default.

mikailweston, 2015-02-18:
I agree, I recently had the problem that I was unable to compile using
VS2013 as I had not updated the eCube API and uninstalled the previous
version.

If the API will be updated frequently it may be better to deactivate the
compile flag and put instructions in the readme for how to install the
eCube API and reactivate the compile flag.

mborisov1, 2015-02-18:
Ok, I'll deactivate the ECUBE_COMPILE flag in the project file the next time I'm going to commit to the repository.

mborisov1, 2015-03-24:
Seems that the ECUBE_COMPILE flag is gone from the VS2013 project file(s). I'm not going to commit the project files with this option set anymore.

closed 2015-03-24 by mborisov1