open-ephys/GUI #255: impedance measures crashes when save to file option is enabled.

shayo, 2015-02-24:
Master branch:impedance measures crashes when save to file option is enabled.

BTW, what happened to my implementation that also shows in GUI the impedance results?
Can you only see the values after they are saved to disk?

aacuevas, 2015-02-24:
This is the same as #246 and related to #253 . Recent changes broke half the FPGA canvas implementation, impedance measurements included. As I've introduced yet more changes in the rhd2164 branch to better support 64channel headstages, I'm waiting for that branch to be fully tested and merged back before addressing this, as it might break again when merging that branch.

jsiegle, 2022-07-30:
The XML file containing impedance values is now saved directly from the Acquisition Board plugin, rather than at the start of recording.

closed 2022-07-30 by jsiegle