open-ephys/GUI #256: DAC audio is incorrect on master branch

shayo, 2015-02-25:
Something doesn't work right with the audio output (via DAC) in the master branch.
It doesn't output the correct channel. Not sure exactly what it does output.
I ran a couple of experiments recently, and it I noticed right away that something is weird....Please check.
It works well in the old spike sorting branch.

aacuevas, 2015-02-25:
Do you mean selecting the output from the FPGA editor, or from somewhere else?

shayo, 2015-02-26:
In SpikeSorter.

aacuevas, 2015-02-27:
Ok, I know where the problem is. When we switched to the multisample rate architecture we streamlined some code, which in turn broke some other. This will be fixed along with all the other issues with the Rhythm node, I'm working on it right now.

jsiegle, 2022-07-30:
The source code for the Rhythm Plugins (Acquisition Board and Intan RHD USB) now live here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing any problems with these plugins, please open a new issue in that repository.

closed 2022-07-30 by jsiegle