open-ephys/GUI #258: Feature request: LFP viewer channel spacing

nikolaskaralis, 2015-03-01:
It would be really useful if the user could modify the channel spacing (distance) in the LFP viewer.
This is different from the the spread. When the amplitude of the signals is high (e.g. 250uV voltage range), then the signals of different channels overlap in the viewer, not allowing the viewing of the signal in full detail.

jvoigts, 2016-02-19:
I think this is addressed in https://github.com/jvoigts/plugin-GUI/commit/c57f85d1eca2a0196272481953af0e6a99b027f3 but this is still in my lfp plot refactoring branch and we will need to do some more work and thoroughly test this before we can move it to the main branch. If you are up for testing the branch and submitting issues that would be useful - im pretty sure you can still make this crash fairly easily.

jvoigts, 2016-04-25:
should now be in lfpviewer beta in main repo testing - see if this works for you and let me know if you have suggestions of what else to add.

closed 2016-04-25 by jvoigts