open-ephys/GUI #259: LFP viewer display bug

nikolaskaralis, 2015-03-02:
When recording for ~2 hours, the display of the LFP viewer starts looking weird for some of the channels (it starts with channel 1 and as time passes, it starts happening for channel 2 etc).
The signal starts looking interrupted, however looking at the recorded data, everything looks fine.
Stopping the acquisition and starting again fixes this issue.

aacuevas, 2015-03-04:
This might be tricky to check, but it probably stems from some mismatch at some buffer, I'll look at them. In the meanwhile, if it happens to you again, could you please attach a screenshot to see what it looks like?

nikolaskaralis, 2015-03-04:
Will do.
However it happens invariably every time after little less than 2 hours of
recording.
From what I have noticed, it doesn't happen during acquisition, no matter
how long it is, but only during recording.

aacuevas, 2015-03-04:
Which recording format are you using? Have you checked if it happens with both? Perhaps there is something there that is lagging the GUI when the files get big enough.

nikolaskaralis, 2015-03-04:
I am recording with the OpenEphys format. Never tried with the Kwik one.

One thing that I noticed after the initial post: If I scroll vertically the
LFP viewer it temporarily displays everything in the signal, but then
immediately reverts back to the broken view.

Until I provide the screenshot: The broken view looks as if it was plotted
with dashed line instead of continuous line. And it is not on all the
traces, but only some of the channels (starting from the first).

nikolaskaralis, 2015-03-05:
It seems it is happen during just acquisition as well (not only during recording).

nikolaskaralis, 2015-03-06:
[image]

jsiegle, 2015-03-06:
Thanks for posting that! I've seen something similar as well. Its likely caused by a drift in display buffer alignment over time. So it's a display bug, not something wrong with the underlying data. Aarón, can you add a check to make sure this doesn't happen?

aacuevas, 2015-03-06:
I've seen something like that when mixing different sources, but it is weird to have alignment issues when everything comes from the same place at the same pace. I'll take a good look at the buffers, there must be something amiss there.

nikolaskaralis, 2015-03-06:
As I wrote above, it happens even with a 16ch headstage, and invariably at
some time after ~1h of recording.

aacuevas, 2015-03-08:
I've added a mutex to prevent different threads writing to the buffer indexes. I've had the GUI running for 2h and didn't get that. There were occasional gaps in all channels (missed paint calls due to timing) but nothing so endemic. Does it get better for you?

nikolaskaralis, 2015-03-09:
It seems this fixed it.  I recorded for ~5h and there was no such issue.

closed 2015-03-09 by nikolaskaralis