open-ephys/GUI #260: 16 channel headstage channel mapping

nikolaskaralis, 2015-03-03:
When the 16 channel option is selected in the Rhythm module, the channels are mapped wrongly (only ~half are correct).

aacuevas, 2015-03-04:
What do you mean with worngly mapped? They appear out of order or with incorrect data? Is consistent or random? If the former, which channels are affected?

nikolaskaralis, 2015-03-04:
I mean that if I set it to 32 channels, I can know which 16 channels are
connected to the ground in the headstage and which 16 channels are
connected to the animal.

However, if I set it to 16 channels, only 8 channels show normal signal and
the other 8 seem grounded, which means that they are wrongly mapped (should
be all 16 channels with normal signal).

aacuevas, 2015-03-04:
I've just checked the datasheet and I think I know where the problem is. The 16 channel headstage has inputs 8-23 connected while we're reading channels 0-16 (we're not doing any mapping in the source node, just reading fewer channels). I'll add some exception code for that headstage so it reads the right channels. In the menawhile the workaround is using the 32ch mode and a channel mapper to disable the unused channels.

nikolaskaralis, 2015-03-04:
Yes, that is what I am doing so far.
Thanks!

Darilbii, 2019-11-20:
This is a old issue, but has this issue been resolved? If yes, how? 

Specifically, how are the Physical Pins of a 16-channel Headstage mapped to the 16 channel output of the Rhythm module of the Gui?

jonnew, 2019-11-20:
I'm fairly sure its just the pin numbers on the intan chip

The electrodes thos are connected to are going to depend on your headstage
and EIB.

closed 2022-07-30 by jsiegle