open-ephys/GUI #262: empty PSTH display

mmyros, 2015-03-04:
Hi all, 

PSTH in the old spikesorting branch still works for me. However, when I tried to use it in the master branch, I got all black subplots instead of PSTH. Trials keep counting. Here is a dump from the command line, I did not find anything useful there. Gdb did not give any clues either. My minimal configuration consists of FPGA -> Bandpass filter -> Spike sorter -> PSTH. Spike sorter has 4 tetrodes, one of them has a unit created from a box. This is all using 2 headstages picking up just ambient 60 cycle noise.

```
m@brain4:~/oe341/GUI/Builds/Linux/build$ ./open-ephys 
JUCE v3.0.6

Created processor graph.

Got audio device.

Audio device name: Default ALSA Output
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created message center.

Loading window bounds.

Item dropped at insertion point 0
Sources::Rhythm FPGA
Creating a new data source.
creating source node.
/home/m/oe341/GUI/Builds/Linux/build
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6010LX45 with serial number 1328000650

FPGA system clock: 100 MHz
Opal Kelly device firmware version: 3.1
Opal Kelly device serial number: 1328000650:�;
Opal Kelly device ID string: Opal Kelly XEM6010

Rhythm configuration file successfully loaded.  Rhythm version number: 1

Initializing acquisition board.
Sample rate set to 30000
Sample rate set to 30000
Number of enabled data streams: 8
Checking for connected amplifier chips...
Number of enabled data streams: 2
Sample rate set to 30000
  Adding node to graph with ID number 100

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 70
Setting buffer address to 0x22db090
Input buffer address is 0x22db090
Item dropped at insertion point 1
Filters::Spike Sorter
Creating a new spike sorter.
  Adding node to graph with ID number 101

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 70
Spike Sorter updating settings.
Item dropped at insertion point 1
Filters::Bandpass Filter
Creating a new filter.
  Adding node to graph with ID number 102

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 70
Bandpass Filter updating settings.
Spike Sorter updating settings.
Item dropped at insertion point 3
Sinks::PSTH
Creating a PSTH output node.
  Adding node to graph with ID number 103

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 70
Bandpass Filter updating settings.
Spike Sorter updating settings.
PSTH updating settings.
Setting hardware trigger alignment channel to -1
Updating SpikeSorterCanvas
Updating SpikeSorterCanvas
Added electrode (ID 1) with 4 channels.
  Channel 0 = 0
  Channel 1 = 1
  Channel 2 = 2
  Channel 3 = 3
Added electrode (ID 2) with 4 channels.
  Channel 0 = 4
  Channel 1 = 5
  Channel 2 = 6
  Channel 3 = 7
Added electrode (ID 3) with 4 channels.
  Channel 0 = 8
  Channel 1 = 9
  Channel 2 = 10
  Channel 3 = 11
Added electrode (ID 4) with 4 channels.
  Channel 0 = 12
  Channel 1 = 13
  Channel 2 = 14
  Channel 3 = 15
Updating SpikeSorterCanvas
Adding new spike plot.
Updating SpikeSorterCanvas
Adding new spike plot.
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 70
Bandpass Filter updating settings.
Spike Sorter updating settings.
Updating SpikeSorterCanvas
Adding new spike plot.
PSTH updating settings.
Updating SpikeSorterCanvas
Adding new spike plot.
Adding tab with index 3
Updating connections:

Signal chain: 0

Source node: Rhythm FPGA.
     Connecting to audio and record nodes.
     Connecting Rhythm FPGA 100 to Bandpass Filter 102

Source node: Bandpass Filter.
     Connecting to audio and record nodes.
     Connecting Bandpass Filter 102 to Spike Sorter 101

Source node: Spike Sorter.
     Connecting to audio and record nodes.
     Connecting Spike Sorter 101 to PSTH 103

Source node: PSTH.
     NOT connecting to audio and record nodes.
     No dest node.

Enabling processors...
Source node received enable signal
Expecting 70 channels.
Number of 16-bit words in FIFO: 0
Is eval board running: 0
Starting acquisition.
Flushing FIFO.
   Enabling VisualizerEditor
SpikeSorterCanvas beginning animation.
PeriStimulusTimeHistogramNode::enable()
   Enabling VisualizerEditor
PeriStimulusTimeHistogramCanvas starting animation.
Filter Viewport disabled.

Adding audio callback.
Adding tab with index 4
Adding box unit to electrode 4
Adding new box unit.
PSTH node received: unitintervalstart
PSTH node received: unitintervalstop

Removing audio callback.
Disabling processors...
Disabling Record Node
Disabling Audio Node
Disabling Message Center
Disabling Rhythm FPGA
Source node received disable signal
RHD2000 data thread stopping acquisition.
Thread exited.
Flushing FIFO.
Number of 16-bit words in FIFO: 0
SourceNode returning true.
Disabling Spike Sorter
Disabling Bandpass Filter
Disabling PSTH
PeriStimulusTimeHistogramNode disabled!
PeriStimulusTimeHistogramCanvas ending animation.
Filter Viewport enabled.
```

aacuevas, 2015-03-04:
The PSTH node was merged into the master branch briefly before we merged the multiple sample rate one. Thus, it still uses the old single-rate sample processing architecture, which explains why it doesn't work now. We'll transition it to the new architecture, altough it might take a while.

jsiegle, 2015-03-06:
The PSTH actually is working in the master branch (although it still needs some cleanup). I'm able to see triggered averages when I use it with the Network Node and the matlab_zeroMQ_wrapper_example.m code in the Resources/Matlab folder. How are you generating your events? Are they coming in as TTLs? It's possible that aspect of the PSTH needs fixing.

[image]

mmyros, 2015-03-06:
Yes, they are coming in as TTLs. I can try the Network Node tomorrow.

mmyros, 2015-03-11:
Network node is not working for me, unfortunately. I followed steps on the wiki to install zeromq, recompiled, reproduced the workflow Josh outlined, and ran matlab example. Trial count stays at zero. I'm not sure where to go from here. Also, my event file got corrupted when I used TTL triggering. I think something is very wrong with my event setup.

aacuevas, 2015-03-11:
Be sure to disable any kinf of saving on the PSTH node. I'm moving the code to the new recording framework, but right now it corrupts the event file.

shayo, 2015-03-11:
Or... Add the Record Control module, which will make sure PSTH dumps things
to disk :)

mmyros, 2015-03-11:
Aarón that must be why. Do you think it's possible to un-corrupt the event files? I have a few that got corrupted during an experiment.

mmyros, 2015-03-16:
Aaron it looks like .spike files are also corrupted (see https://github.com/open-ephys/GUI/issues/146#issuecomment-81966080)

aacuevas, 2015-03-17:
The spike issue in #146 seems related to the change in format for spike files instead of a file corrption. The simple way to know if a file is actually corrupted or not is to try to load it with our load_open_ephys_data.m MATLAB function, which is updated to the latest format version.

I updated last week the PSTH node to use the new record subsystem, so all new files should be well.

@mmyros, send me your files to aaron@open-ephys.org and I'll see if they can be fixed or not (it depends on how the data was written), we can talk over there.

closed 2015-05-09 by mmyros