open-ephys/GUI #265: Cannot raise the second threshold (red line) in the Spike Detector

shinkira, 2015-03-12:
By default, the first threshold (orange line) is set at 50 and the second threshold (red line) is set at zero, which is indicated at its left edge (please see the attached image). In the Spike Viewer window, I can move the second threshold up and down, but cannot (visually) raise it above the default height and the value of the second threshold never goes below zero. Is this a bug, or is there a way to raise the second threshold?

[image]

shinkira, 2015-03-12:
I followed the tutorial, and this is the signal chain:

[image]

koreign, 2015-03-20:
I have the same problem. Actually if you use the spike sorter module instead of the spike detector, you can move the threshold from -inf to +inf, but it will stop at 0 in the spike viewer. Also int he spike sorter, spike appear already inverted therefore the threshold works well, but not with spike sorter+spike viewer. Not sure why.

shinkira, 2015-03-20:
Thanks for the comment. It is good to know that the problem is reproducible in other environments. Yes, I noticed that the Spike Sorter works better than Spike Viewer. I guess I can use the Spike Sorter in the mean time, but I do not know how much I can trust the software when it does not even work as described in its tutorial...

jsiegle, 2015-03-20:
This is not a bug. The first threshold (orange) indicates the threshold set in the Spike Detector, and is based on information in the spike events that are emitted by that processor. The second threshold (red) indicates the "display" threshold, which is used to filter out low-amplitude spikes from being visualized or saved. If it's below the first threshold, it will have no effect, which is why it doesn't need to be able to cross the zero line. But note that if you choose the "invert spikes" option, the threshold will switch to appearing above the zero line.

In general, we recommend that people use the Spike Sorter, rather than the Spike Detector, for spike extraction, but we've kept it around so that signal chains that depend on the Spike Detector can still be used.

@shinkira, can you point out the aspect of the tutorial that isn't working for you?

shinkira, 2015-03-20:
Here is the part of the tutorial that I could not follow:

"Click on the red line of one of the waveform plots in the Spike Viewer and drag it upward [***** THIS IS IMPOSSIBLE AT ITS DEFAULT HEIGHT (= 0) *****]. When the line is at zero, all the spikes from the Spike Detector will be displayed. As you move it higher, more and more spikes will be filtered out based on their peaks."

In the image of the Spike Detector below, please note that narrow spikes (green arrow) are contaminated by wide spikes (blue arrow). If I can raise the threshold (red line) above up to the magenta dashed line, I can filter out the the wide spikes. But the GUI does not allow me to raise the threshold. When I choose to display inverted spikes, everything including the threshold setting gets inverted. It is similar to just flipping the display upside down. Consequently, I still cannot filter out the wide spikes.  

[image]

jsiegle, 2015-03-20:
The tutorial was written before non-inverted spikes became the default setting for the Spike Viewer. I just changed it to say "downward" and "lower" instead.

Those examples looks like artifacts, not spikes. Note that they are pointing in the wrong direction for typical spike waveforms. It's not impossible to observe actual spikes that appear inverted, but conventionally they are simply ignored.

shinkira, 2015-03-20:
Then, I think experimenters want to filter out the artifacts in the range where the signal-to-noise ratio is at its maximum—that is, in the range of magenta line rather than than of the orange line. Right now, this seems impossible to do.

jsiegle, 2015-03-20:
If I understand correctly, this would involve having a second threshold to filter out positive-going events above a certain level. This is possible to implement, but seems like something that would be more practical to do offline. Also, with the Spike Sorter, you can use box windows to achieve the same result.

shinkira, 2015-03-20:
Yes, I think I should use the Spike Sorter for that purpose. Thank you for the suggestions and updating the tutorial.

closed 2022-07-30 by jsiegle