open-ephys/GUI #269: Issue with Channel Mapping and Reference bank

koreign, 2015-03-19:
Dear All,

    Thank you again for the great work. I noticed a quite disturbing problem while playing with the channel mapping module:

The module is used for:
1. reordering channels
2. activate deactivate channels
3. organize reference banks

     The issue:
1. I used the simple chain: file reader (with provided cortext rec) -> channel map -> lfp viewer
2. in channel mapping: i reordered channel No-5 to the first position (and in the param tab 5 also shifted to first position) - nothing wrong so far
3. I selected the bank A and added the 4 first channels (5 1 2 3).
4. now I chose the reference to be channel No-5 and magically channel N0-5 was not 0 (chan1 in lfp viewer) but the second channel in the lfp viewer which should correspond to channel 1 after remapping.
 
    Now this is both confusing and worrysome. I am not sure if it is something I am doing wrong but definitively, for people doing cross referencing and all, this will induce huge problem.

Koreign

aacuevas, 2015-03-20:
It is definitely a bug, probably derived from how we changed some underlying methods to be able to support multiple sources with different sample rates, some index must have been moved, I'll look at it asap.

To know the version of the GUI, you can look at the bottom right corner of the "Graph" visualizer tab.

aacuevas, 2015-03-20:
Should be fixed now in the latest version of the source code.

koreign, 2015-03-21:
Thanks. I still have some trouble with compiling so I will have to wait until the next binary release for now.

closed 2015-03-21 by koreign