open-ephys/GUI #270: Crash when deleting a source node

mborisov1, 2015-03-24:
I've noticed that recent commits of OpenEphys/master crash when the user deletes a source node in the GUI. The crash also happens when the user closes the program when there were source node(s) added.

I've traced the problem to the file ProcessorGraph.cpp, function "removeProcessor". There is some kind of iteration on processors, statically downcasting them to the type "GenericProcessor". However, for some reason, one of the processors can't be downcasted to GenericProcessor. This was causing the crash. I've changed it to a dynamic_cast that fixes that fixes the crash, but I'm not sure if it's ok that one of the processors is not of the type "GenericProcessor". Please somebody who deeply understands the program architecture check it to see if there are more bugs related to my finding.

aacuevas, 2015-03-24:
I think I know where the issue was. All the processors are indeed derived from GenericProcessor except for for the one that sends data to the audio hardware, which is a juce internal one. I find weird that we didn't got any errors earlier because of this, good catch.
I'll add some checks to that function, just to be on the safe side.

mborisov1, 2015-03-24:
Thanks. And please also check other cases of static_cast<GenericProcessor*> that may also lead to crashes in different circumstances.

aacuevas, 2015-03-24:
There shouldn't be any, at least not like this one. Outside of ProcessorGraph all we ever use are our own processors (which are all derived from GenericProcessor) and I've checked that all other casting instances in ProcessorGraph have this check in place.
I've left the dynamic_cast for the time being, though, it shouldn't hurt.

closed 2022-07-30 by jsiegle