open-ephys/GUI #271: Improper assignment AUX/ADC channels to single vs multiple analog inputs

linussun, 2015-03-25:
I am getting a strange behavior on how GUI activates ADC and AUX channels. When I plug in a single analog input into channel 1 analog input board: Instead of the FPGA module ADC button activating the ADC channels 1-8, only ***AUX*** channels 1-8 are active, the ADC do not appear. The signal appears as an AUX signal and is 1000x less in voltage (2.5V signal becomes 2.5 uV) and is adjustable as an AUX 'range' signal, not an ADC range of voltages signal. 
2. If I plug in 3 analog input channels to channels 1,2,3 or any other combination say channels 2,3,4, the FPGA module will activate AUX1 and ADC2-8 in the FPGA tab in GUI. It seems as if AUX1 is always on no matter what is connected to it. Even if there are no inputs to the BNC connectors, as long as the HDMI cable is connected, AUX1 is active.

aacuevas, 2015-03-25:
What do you mean with "active" ? AUX1-3 are always present, as they are the auxiliary inputs of the headstage (not the board), usually connected to an accelerometer. ADC inputs are manually enabled by the corresponding button, as the board and the software have no way of knowing if a IO board is plugged or what channels are connected.
What you describe seems like a bug on channel type assignments. Are you using the latest code from github or the precompiled binaries? Could you please attach some screenshots showing the issue? Are the channels also wrongly mapped in the saved files?

linussun, 2015-03-26:
Hi Aaron,

I didn't know Aux1-3 are always present. If I only have an HDMI cable connected (and not a 32 channel intan headstage+cable to port A) to the acquisition board: when starting up the FPGA module and adding the LFP viewer next, only Aux 1 shows up. If I press the ADC button on the FPGA module, then AUX1-8 is diplayed on the LFP viewer. I've attached screenshots of this effect. Even though the only input I have is via ADC channel 1 or 2 (I tried at least), the display shows Aux2-8 which are displayed according to the AUX settings, plus  my 2.5V signal appears as 2.5 uV according to ranges set by AUX channels.

When I attach more than one analog input to the passive analog input board, then ADC2-8 appears in the LFP viewer, and AUX 1 remains (no screenshot of this yet). I've attached screenshots of the first scenario. On ADC2, the 2.5V signal appears correctly. I haven't examined the files yet.

Thanks,

Linus

______________________________________
Linus Da-Shih Sun, Ph.D., M.D.
Instructor of Neurology in Ophthalmology
Columbia University
1051 Riverside Drive, Unit 87
Kolb Research Annex Rm. 561
New York, N.Y. 10032
Phone: (212) 543-6931 ext. 605
Fax: (866) 683-9592
email: ls2747@columbia.edu

aacuevas, 2015-03-26:
Ah, you aren't using any headstage, that changes things. Every headstage always has 3 aux channels, but that isn't the case with no headstage connected. When no headstage is plugged and with the ADC disabled (it does not matter if the HDMI is connected or not) the FPGA is reporting 0 channels but the software inserts one dummy channel. We should add an extra check to avoid being able to start acquisition in this case.
What is happening is that when activating the ADC channels with no headstage, the FPGA module doesn't assign the output channels properly. It's an uncommon scenario and we hadn't tested much with no headstage, but knowing it now I'll work on fixing it.

linussun, 2015-03-26:
Ah, I see, okay I'll plug in the headstage and see what happens if that clears it up. BTW I would suggest that instead of preventing acquisition if headstage is removed, you might want to still allow acquisition and let GUI be more flexible as to what information it can store (e.g. eye position and behavior only is our interest for some pure behavioral studies or the in future someone might want to store video frames from a two photon image etc...) this allows for testing data files without dummy headstage data too.

Thanks

______________________________________
Linus Da-Shih Sun, Ph.D., M.D.
Instructor of Neurology in Ophthalmology
Columbia University
1051 Riverside Drive, Unit 87
Kolb Research Annex Rm. 561
New York, N.Y. 10032
Phone: (212) 543-6931 ext. 605
Fax: (866) 683-9592
email: ls2747@columbia.edu

aacuevas, 2015-03-26:
The idea wasn't to prevent acquisition if no headstage is present, but if a rhythm node reports 0 channels (i.e.: a source node with only adc channels would still be able to acquire). It might be possible to make it able to work with only event data and no data channel whatsoever, but it would require some deep changes, as by design everything revolves around the signal chain for continuous data.
You can, however, select which channels are stored and which not (even not saving any continuous data).

closed 2022-07-30 by jsiegle