open-ephys/GUI #272: PSTHs not updating, though trial count is.

dmschneid, 2015-03-26:
I am trying to create PSTHs triggered off on an external TTL. My data are coming in through an Intan system. I am sorting a single unit on Ch1 (which is actually just 60Hz noise, but produces many "spikes" per second). A single PSTH window shows up in the PSTH pane, as it should, and the trial counter increments appropriately every time I send a TTL, but the traces remain flatlined at 0 Hz, despite that fact that my sorted unit is "spiking" at 10s of Hz. Any thoughts?

As an aside, I noticed that even if I don't select a TTL for the trigger, my trial counter still increments every time I send a high signal to TTL1.

Thanks,
David

[image]

jsiegle, 2015-04-03:
You shouldn't see flat traces anymore...the spikes are now being aligned correctly. I would still recommend testing it with some known inputs, though, in order to validate the temporal alignment.

jsiegle, 2022-07-30:
The PSTH plugin has been substantially refactored and now lives here: https://github.com/open-ephys-plugins/online-psth

If you're having any problems with that plugin, please open an issue in that repository.

closed 2022-07-30 by jsiegle