open-ephys/GUI #273: cannot find libokFrontPanel.so or rhd2000.bit when loading Rhythm FPGA module

cxrodgers, 2015-04-15:
I just updated to the newest version of the GUI on git and I'm getting a couple of new bugs when loading the Rhythm FPGA module. I'm using Linux Mint 17 x64, and I'm running it from the Builds/Linux/build directory.

First, after I drag the Rhythm FPGA module to the signal chain, I get a popup that says "Opal Kelly library not found. The Opal Kelly library file was not found in the directory of the executable." From the terminal, I can see an additional error message: 

./libokFrontPanel.so: cannot open shared object file: No such file or directory
FrontPanel DLL could not be loaded.  Make sure this DLL is in the application start directory.

If I use the dialog to navigate to Resources/DLLs/libokFrontPanel.so (or libokFrontPanel64.so), then I just get the same error message again.What I ended up doing was copy the libokFrontPanel64.so into my current directory (Builds/Linux/build), and then rename it libokFrontPanel.so. Then it finds it perfectly well.

Next, I get a popup saying "FPGA bitfile not found. The rhd2000.bit file was not found in the directory of the executable." I choose GUI/Resources/Bitfiles/rhd2000.bit from the dialog box and from then on everything runs normally.

So, this workaround gets the job done, but it would be nice if it could find these files normally, which I am pretty sure it used to do in the version from git a couple of weeks ago.

Also, slightly unrelated, but whenever a File Selector dialog box is opened (as above, or when selecting a path to save recordings), it's always behind all of the current windows, and it doesn't show up in the menu bar. So, it seems like the software has hung, but really it's just waiting for user input in a hidden window. Is that a known issue? Should I open a new issue for that? Thanks!

aacuevas, 2015-04-15:
The DLL selection dialog should work now, without having to move the files to the binary directory (although you can still move it as well as the bitfile for the GUI to automatically load them).

All the file selection dialogs are set to use the system's native file chooser if possible, so I'm not sure if we can do anything about the window priority or menu bar.

cxrodgers, 2015-04-17:
Ah, I see what happened. When I recompiled, there was some error, so I did a "rm -rf build/" to get back to a clean state. I didn't realize that there are supposed to be some files (libokfrontpanel, and rhd2000) in the build directory. My fault. Thanks for fixing the dll chooser anyway though!

About the window priority, there is just something funky happening where it opens the dialog behind the current window. This doesn't happen to me with any other application but it could be some peculiarity of my system, I suppose. If I find a way to replicate it, I'll post it as its own issue.

closed 2015-04-17 by cxrodgers