open-ephys/GUI #274: NetworkEvents: how to sync with continuous ADC data?

linussun, 2015-04-17:
When using matlab to extract messages.events files, the time stamps are in integers and appear to be called by t.getHighResolutionTicks().  However ADC continuous data timestamps are in seconds.  Is there any way to convert easily between the two?  I've tried dividing by the SR=30000 but it doesn't appear to sync my nework events with my photodiode (screen).

Question: are the timestamps of the continuous channels sampled from the clock on the FPGA? are the timestamps from NetworkEvents sampled from the same clock on the FPGA?

It would be perfect if the network event message timestamps were in seconds like the continuous channels, or vice versa, to synchronize them to the best of the hardware's capabilities (I realize there may be a few ms difference due to delays in TCP/IP network delays/handling).

shayo, 2015-04-17:
Aligning software to hardware timestamp is always an annoyance.
There is no hardware time stamping of network events. You need to sync two
separate clocks (FPGA, CPU). In some systems, slight differences in the
quartz timing can lead to significant drift over time.

What I ended up doing is keeping both and then fitting a linear model that
map time stamps in one time zone to another. This works quite well.
You can look in load_events_version_0_3 (strctHardwareSoftwareSync)

linussun, 2015-04-22:
There is a "time elapsed" displayed in GUI, what time is that timer synced to, the hardware of software? How difficult would it be to poll the hardware timestamp for network events? At least it will be on the order of milliseconds and would be helpful.  

Also noticing from your code that you are doing a linear regression for spikes too, why is this, shouldn't the spikes be using the hardware timestamp?  

I think in my setup, where I am not sending eye position every few milliseconds by network events, but instead sending eye position via an ADC continuous channel (mVolts). My network events are not synced to any reasonable continuous signal yet (e.g. by a 'pulse step') -thus a reasonable linear model (within a few ms) doesn't seem feasible. This will be something I will need to figure out a solution for, and makes network events a pain to deal with unless it can do a reasonable poll of the hardware clock.

shayo, 2015-04-22:
The only way I know to get a somewhat reasonable software-hardware sync is
to query the software clock as soon as you get a hardware timestamp (I'll
be happy if someone here has an alternative method how to do this better).

This is why whenever a USB packet arrives, it is also software timestampped
(or at least, the timestamp of the first sample is estimated by subtracting
the sample rate times how many samples arrived).

Network events arrive asynchronisly and are software timestamped. This
software timestamp can be converted to hardware timestamp by searching for
the two closest usb packets, taking their software and hardware timestamps
and interpolating between them. A somewhat more robust approach is to use
linear regression. This reduces jitter since there is no guarantee that
your query for a software timestamp will be served immediately.

Of course, it is always recommended to send a TTL pulse that is locked to
your stimulus onset and align everything to the hardware clock, then you
only infer which stimulus was delivered from the network events (and jitter
of ~3 ms shouldn't affect anythign).

Spikes are both hardware and software timestampped just for convenience
(essentially, the software TS is estimated from the hardware by using a
similar linear regression approach).

linussun, 2015-04-24:
I think this is reasonable to do interpolation between USB packets (how often are they sent?) with a linear regression model. As long as the network events sent by the behavioral/display control fall within a video frame refresh or two prior to the rendering on the actual display and captured by a photodiode input into an ADC channel, this should be adequate for inferring the state of the trial.

jsiegle, 2015-04-29:
If everything is working correctly, the messages.events file should be using hardware timestamps. This includes network events, which are given a hardware timestamp (rounded to the nearest buffer start, on the order of ~25 ms) when they arrive. Much of the internal processing within the PSTH node uses software timestamps, but messages.events will always have hardware timestamps. The first event includes the software timestamp when recording was started, in case you need it for alignment purposes.

You mentioned the network timestamps aren't properly aligned when you divide by the sample rate. How far off are they? I just tested it on my machine, and the numbers make sense.

There is a problem with the software timestamp being saved in binary at the end of each network message, but you can just ignore that. I'll open a separate issue about it.

linussun, 2015-05-05:
I believe I figured out the issue of mis-alignment of my setup, I was using a virtual machine, and the display of the fast moving LFP signal in GUI in vncviewer was slowing down the VM to the point of slowing down USB packet arrival times to GUI, thus causing the network events to have an 'earlier' timestamp extracted from the USB packet that arrived late in time.  Running GUI natively helped me align network events and photodiode events, evidenced here where the continuous signal is an ADC input for a photodiode probe and network events plotted by timestamp. It makes sense now, this link is to a matlab plot of network events to photodiode events: 

https://dl.dropboxusercontent.com/u/26686593/OEP/networkevents%20aligned%20with%20photodiode.tif

These network "hardware timestamps" work for me as they appear about 1000-1600 'hardware' timestamps before my photodiode signal. Josh, can I confirm with you that these hardware timestamps are actually 'sample numbers' and thus we have to divide by the sample rate to convert to seconds?  Thanks.

To get to an

closed 2020-12-07 by linussun