open-ephys/GUI #275: Managing Analogue Output

koreign, 2015-04-22:
Hi,

  I am trying to use the analogue output to return one of the acquired channel, eventually filtered. I am not sure about:
1. how to chose the channel I want to output
2. can it be done in most module ?
3. how do I chose on which channel its output will be (1-8) ?

aacuevas, 2015-04-23:
What you want to achieve is not possible currently. Right now the analog outus can only have an internal hardware redirection of a raw channel, directly from the headstages. Right now you can set channels 0 and 1 using the Rhythm node, we should definitely add some way to direct channels to the remaining outputs.

For any processor to send data to the hardware analog outputs we'd have to modify the Rhythm firmware on the FPGA, as it doesn't currently allow that. It would also work at the expense of data bandwidth and with no little amount of synchronism issues to solve. If there were enough interest on that feature we could look into that but only after we have USB3.0 boards up and running and only then we'd see if we could make it work on 2.0 as well.

koreign, 2015-04-24:
I tried using the FPGA audio output (which is sufficient for me right now), I sed the minijack output on the board, which return a very low amplitude signal (I was told that it would be the default output if no IO board is plugged in). I will try today with a IO board see if chan 0 and 1 contains the data or not.

Another question is then: what is the use of IO analogue ouput then ? or How do/can we use it?

aacuevas, 2015-04-24:
Analog outputs 0 and 1 are the same that channels L and R of the jack. to use them you have to select the appropiate button under "Audio out" on the Rhythm source node and then select the desired channel on the "param" tab of the processor (click the left side to unfold the channel selector).

All analog outputs can right now do the same: output raw channel data from the headstages, altough only channels 0 and 1 can be selected right now, but that it's merely a graphical interface issue, the functionality is there. We just have to think how to add that to the (actually rather crowded) processor interface.

Other uses (i.e.: sending arbitrary data from the software) are physically possible but would require non trivial changes to both software and firmware.

koreign, 2015-04-24:
I see, thank you for your kind answers.
A second concern is the last time I tried reading the output of the minijack the signal was very low amplitude. I will try again on monday to measure it and will get back to you on that (if i may).

koreign, 2015-05-14:
I tried to test the analogue output using the two methods:
1. in FPGA module -> select left/right and channels with params
2. in FPGA module -> select audio

I am not exxactly sure how the audio works since several channels could be selected and seems to sum up and right now i would be interested in isolated channels.

The option 1 currently returns a signal but I am not sure about the characteristics of the ouput (range, offset etc). on the bare arduino the input range is 0-5V so i need too fit the analog output of OpenEphys in order to read the non truncated signals, does any body know the output characteristics ?

aacuevas, 2015-05-14:
For the software output usualy you should select only one channel but if more are selected, yes, the software sums all inputs.

The hardware audio signal has a -5V to +5V range with zero offset (except for the one caused by the output hardware stage, in the range of mVolts)

closed 2022-07-30 by jsiegle