open-ephys/GUI #276: Record Control

koreign, 2015-04-23:
Hi,

  Sorry again for the stupid questions but I don't get how to use the record control module.

I used a basic configuration (FPGA - channel map - band pass - lfp viewer)(note there are 2intan 16ch attached, a ttl-IN IO board and a analog-out IO board).

I added the record control betweem fpga and the channel mapping. I tried sending the ttl during acquisition (in that case I can see it in lfp viewer but no recording is started), or while no acquisitionis being performed.

So far nothing happens. Could you tell me what I am doing wrong ?

I am on 
window 7 - 32, 
GUI v 0.3.5

aacuevas, 2015-04-23:
How long are your pulses? The record control works on edge, starting recording when it detects a rising edge and stopping when a falling edge is sent. That means that if you're sending small pulses it won't have time to start recording.

shayo, 2015-04-23:
We should add an option to record either with that mode of operation (line
is high), or with two triggers (i.e. oen for start, one for end).

koreign, 2015-04-24:
I used a short pulse at the beginning (50msec), then realize that as  you said it might be slope dependent, so I used a 0.5 sec pulse duration (still nothing happened), then I just turned a constantly ON TTL and still nothing happened. 

I will try again and see if I can reproduce that. I will test it on the two boards I have.

I think it would be nice if it was in the documentation that 
1. it is slope dependent
2. long pulse duration of  >Xmsec are needed to be executed
3. is acuisition required to be running or not etc.

aacuevas, 2015-04-24:
I've tested with short pulses and it works for me, so I'm not sure what's happening in your case. Have you checked you have the right ttl channel selected on the record control node? Can you please save your chain and send me (to aaron@open-ephys.org) so I can look at it? 
Acquisition running is required for basically everything, as all the processors are disabled and do not run anything otherwise.

I agree that having a couple of extra modes of operation should be handy. I'll add them and then udate the wiki to be sure it's clearly explained how to use the node.

koreign, 2015-04-24:
Hi,

   I retested everything and as you mentioned it works fine right now. I must have plugged to the wrong ttl input or selected the wrong ttl trigger. 

1. Why is there 10 channels selectable for the ttl input  and not the 8 accessible from the board ?
2. how do we know the number for internal ttl (such as phase detector or else coming from network event ?)

closed 2022-07-30 by jsiegle