open-ephys/GUI #277: Bug: messages.events timestamping

macstry, 2015-04-23:
I was analyzing my continuous LFP data and trying to line-up manual timestamps that I had placed during recording, when I noticed that clear events on the recording were not lining up with the timestamp. For a number of my recordings, I had hit the play button prior to hitting the record button in the GUI to make sure that I had a good signal before recording. I was wondering whether the timestamp on the messages.events file related to the sample number on the actual recording or the overall sample number (including non-recorded time). I did a test where I did a number of timestamps, paused the recording, and did additional timestamps. From that data, it looks like the timestamp includes non-recorded time (which means that the i64 stamp doesn't actually correspond to sample number in the recording.) I have a lot of data that hinges on accurate timestamps and is already recorded- is there any way to determine how these timestamps line up? And can this timestamping system be fixed in an update?

macstry, 2015-04-23:
This should probably actually be in the import section, as I think this is a results of the matlab importer not necessarily importing the same timestamps (?)

closed 2015-04-23 by macstry

linussun, 2015-04-28:
Take a look at #274 
https://github.com/open-ephys/GUI/issues/274#issuecomment-95326490

I am having the same issue, and we have discussed a solution which involves a linear regression bewteen timed events, but note that messages.events is NOT synchronized by hardware to the FPGA hardware clock which records timestamps with continuous LFP data.