open-ephys/GUI #283: Ubuntu 15.04 installation problem

tomiwase, 2015-05-13:
Hi!

In short. I have installed Open ephys and HDF5 libraries with the guide. When I try to run open ephys I get the following error message:

tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ gdb ./open-ephys
GNU gdb (Ubuntu 7.9-1ubuntu1) 7.9
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./open-ephys...done.
(gdb) run
Starting program: /home/tomiwase/Open Ephys/open-ephys 
/home/tomiwase/Open Ephys/open-ephys: error while loading shared libraries: libhdf5.so.7: cannot open shared object file: No such file or directory
[Inferior 1 (process 5700) exited with code 0177]
(gdb) backtrace
No stack.
(gdb) 

Obi-whoeveryouarepleasehelpme you are my only hope.

FloFra, 2015-05-13:
Can you please paste us the output of ```ldd open-ephys```?

Also, please try to run ```LD_LIBRARY_PATH=/usr/local/hdf5/lib ./open-ephys``` and see if that works.

Is there a specific reason why you build open-ephys from source and not use the precompiled binaries?

tomiwase, 2015-05-18:
tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ ldd open-ephys
	linux-vdso.so.1 =>  (0x00007ffe68a26000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f23062b5000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f2305f7c000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f2305d69000)
	libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007f2305a78000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2305874000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f23055cb000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f23053ad000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f23051a5000)
	libhdf5.so.7 => not found
	libhdf5_cpp.so.7 => not found
	libzmq.so.4 => not found
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2304e94000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2304b8c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2304976000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f23045ab000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f2304382000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f2304158000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f2303f54000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f2303d4e000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f2303b4c000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007f2303934000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007f230372f000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007f230352c000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007f2303328000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f2303122000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f2302f03000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007f2302d00000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f2302afa000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f23028ed000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f230656a000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f23026d1000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f23024ab000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f23022a6000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f23020a0000)
tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ 

tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ LD_LIBRARY_PATH=/usr/local/hdf5/lib ./open-ephys
./open-ephys: error while loading shared libraries: libhdf5.so.7: cannot open shared object file: No such file or directory
tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ LD_LIBRARY_PATH=/usr/local/hdf5/lib
tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$  ./open-ephys
./open-ephys: error while loading shared libraries: libhdf5.so.7: cannot open shared object file: No such file or directory
tomiwase@tomiwase-Latitude-E6420:~/Open Ephys$ 

>Is there a specific reason why you build open-ephys from source and not used the precompiled binaries?
Well no. How can I do it?

FloFra, 2015-07-28:
Well, it looks like you ever installed the hdf5 library properly or at least not to ```/usr/local/hdf5/lib/ibhdf5.so.7```.

The precompiled binaries can be found [here](https://github.com/open-ephys-GUI-binaries).

krisalpha, 2016-03-18:
you can refer to installation notes i made for installation on ubuntu 15.04 ,, refer to attachment
[github-283.txt](https://github.com/open-ephys/GUI/files/180361/github-283.txt)

jvoigts, 2016-03-19:
Hi,
For instances where these differ from the existing documentation can you add the installation notes to the wiki? (https://open-ephys.atlassian.net/wiki/display/OEW/Linux)

Just add a note like 'for ubuntu 15.04, do xyz'

krisalpha, 2016-03-19:
sure i will add them soon

closed 2022-07-30 by jsiegle