open-ephys/GUI #284: cannot find rhythm board on Ubuntu 14.04

ericyosho, 2015-05-14:
Hi, 

We successfully build OpenEphys from source code based on the following commit and the tutorial example is running without a problem.

commit d3106052d21ab9d74ccaca59d8cbe4e9618ae63d
Author: Aaron Cuevas Lopez <aacuelo@teleco.upv.es>
Date:   Wed Apr 29 03:24:28 2015 +0200
    Remove binary timestamp at the end of network messages

We then connect the Rhythm acquisition board through a USB port on the computer.
The lsusb command seems to recognize the hardware without the descriptive string
Bus 003 Device 011: ID 151f:002d     (EMPTY description here)

However, when we launch the open-ephys GUI, it complaints about "Acquisition board not found".
And the terminal says "Find 0 Opal Kelly devices connected".

I searched the wiki but there seemed to be no driver discussions.
Do I miss something or where I should look into next?

Thanks!

aacuevas, 2015-05-14:
Have you copied the 40-open-ephys.rules (located in Resources/Scripts) to the udev directory? (steps 4 & 5 of https://open-ephys.atlassian.net/wiki/display/OEW/Linux ). Linux doesn't need a driver but the udev daemon needs this config file for the library to be able to locate the FPGA.

ericyosho, 2015-05-14:
Hi Aaron,

Thanks for the information.
I copied the udev file into the folder, but restarting the udev service
alone does not seem to work.

After I reboot the machine, the GUI can find the device sometimes.
Occasionally, the GUI complains the following error message:

error submitting URB: Cannot allocate memoryerror submitting URB: Cannot
allocate memoryerror submitting URB: Cannot allocate memoryerror submitting
URB: Cannot allocate memoryerror submitting URB: Cannot allocate
memoryerror submitting URB: Cannot allocate memoryerror submitting URB:
Cannot allocate memoryerror submitting URB: Cannot allocate memoryerror
submitting URB: Cannot allocate memoryerror submitting URB: Cannot allocate
memoryerror submitting URB: Cannot allocate memoryerror submitting URB:
Cannot allocate memoryerror submitting URB: Cannot allocate memoryerror
submitting URB: Cannot allocate memoryerror submitting URB: Cannot allocate
memoryerror submitting URB: Cannot allocate memoryerror submitting URB:
Cannot allocate memorySegmentation fault

Best,
Zhe Yao
--------------
Department of Electrical and Computer Engineering
McGill University
Montreal, QC, Canada
H3A 2A7

zhe.yao@mail.mcgill.ca

aacuevas, 2015-05-14:
Hi,

"After I reboot the machine, the GUI can find the device sometimes." Do you mean it is not always recognized? After a URB error appears you'll have to completely unplug the board and plug it again.

URB errors happen when the opal kelly frontpanel library ends in an unexpected state or when there is too much data load for it. Are there any specific steps that reproduce the issue for you?

ericyosho, 2015-05-14:
Hi Aaron,

As far as we can tell, the acquisition board now can be recognized always,
but sometimes the GUI crashes because of the URB error.

When we play with the GUI in Linux, it crashes (meaning no response to the
user input anymore, and cannot be killed using Ctrl+C).
We have to open another Terminal window to kill it, and we can see the CPU
usage is often 100%.
This URB error usually happens if the GUI crashes last time.

Best,
Zhe Yao
--------------
Department of Electrical and Computer Engineering
McGill University
Montreal, QC, Canada
H3A 2A7

zhe.yao@mail.mcgill.ca

jsiegle, 2015-05-14:
URB errors usually happen when the board gets unplugged during acquisition, or if the GUI crashes without resetting the acquisition board properly. Resetting the board before you start the GUI (by unplugging and replugging the power cord) almost always fixes the problem.

closed 2022-07-30 by jsiegle