open-ephys/GUI #285: cannot find opal kelly library file

jennisiegel, 2015-05-18:
Hi everyone!
I am just now finally updating to the newest version (0.3.5) on my mac OS X running yosemite.
When I drop the FGPA into the signal chain I receive the error "cannot find kelly opal library". The front panel library file is there in the same directory (as it came with the download), and even if I choose it manually the board is not recognized. I am using the precompiled version of the GUI. I have not tried building it from the source code at this point.
Has anyone else had this issue?

open-ephys, 2015-05-19:
Hmmm...I was having the same problem with the downloaded executable, but I re-compiled it and it went away. Try downloading it again and see if it works.

jennisiegel, 2015-05-19:
I will do that this afternoon and let you know, thanks so much!

Jennifer J Siegel, PhD
Center for Learning & Memory
University of Texas at Austin

jennisiegel, 2015-05-19:
Success!
Thanks SO MUCH!

Jennifer J Siegel, PhD
Center for Learning & Memory
University of Texas at Austin

aacuevas, 2015-05-19:
Yeah, there was a bug with the DLL selection dialog that was fixed on the source a while ago. We shold update the binaries soon, though.
Glad it is working for you now.

closed 2015-05-19 by aacuevas