open-ephys/GUI #293: serial input

m4loman, 2015-06-20:
Hi there,

I am having trouble getting the serial input tool to work. As soon as I send data the GUI crashes.

Also, an option to use an 8n1 scheme would be nice. Use 1 serial line as a strobe bit triggering on the rising or falling edge and 8 bit to code for events.

All the best

Jan

FloFra, 2015-10-15:
Crashes should not happen, that is for sure a critical bug. 

Can you give me some infos on how to reproduce this issue?

command-paul, 2016-03-12:
Experienced the same issue today, Trying to pass in serial data from a micro controller of mine

rwd122, 2017-09-23:
I am having the same issue. GUI crashes within 1 second when I send input (from arduino) in. I am using "serial port" as the source and "LFP Viewer" as the sink. Windows 10.

FlorianFranzen, 2017-09-24:
Is it crashing when you just run the Serial Input Node or only after you add a sink behind it. Can you supply me with a crash log or even better, a backtrace, best from an OpenEphys compiled with Debug Symbols?

JayPBee, 2019-04-29:
Hi,
I have a somewhat same issue. SerialInput, LFP_Plotter, crash.
The Crash report is below.

However, I can not find anywhere the information, what the serial stream should look like. I have my own microcontroller there and can do whatever you like. 

**### What is the data format for the serial input plugin ??**

Cheers, Joachim

`Process:               open-ephys [1025]
Path:                  /Volumes/VOLUME/*/open-ephys.app/Contents/MacOS/open-ephys
Identifier:            org.open-ephys.gui
Version:               0.4.4.1 (0.4.4.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           open-ephys [1025]
User ID:               501

Date/Time:             2019-04-29 12:36:26.782 +0200
OS Version:            Mac OS X 10.14.4 (18E226)
Report Version:        12
Bridge OS Version:     3.4 (16P4507)
Anonymous UUID:        9BD8B26F-12AF-5F75-8303-71F913B39276

Sleep/Wake UUID:       45839C37-30C2-461B-9C11-5CCC12F73BF1

Time Awake Since Boot: 5700 seconds
Time Since Wake:       5200 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        1  com.apple.audio.IOThread.client

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000002a
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1025]

VM Regions Near 0x2a:
--> 
    __TEXT                 00000001026af000-0000000102b6c000 [ 4852K] r-x/rwx SM=COW  /var/folders/*/open-ephys.app/Contents/MacOS/open-ephys

Thread 0:: Juce Message Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff76bfe22a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff76bfe76c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff4a65113e __CFRunLoopServiceMachPort + 328
3   com.apple.CoreFoundation      	0x00007fff4a6506ac __CFRunLoopRun + 1612
4   com.apple.CoreFoundation      	0x00007fff4a64fe0e CFRunLoopRunSpecific + 455
5   com.apple.HIToolbox           	0x00007fff4993c9db RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff4993c715 ReceiveNextEventCommon + 603
7   com.apple.HIToolbox           	0x00007fff4993c4a6 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff47cd6ffb _DPSNextEvent + 965
9   com.apple.AppKit              	0x00007fff47cd5d93 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
10  com.apple.AppKit              	0x00007fff47ccfeb0 -[NSApplication run] + 699
11  org.open-ephys.gui            	0x00000001028531ef juce::JUCEApplicationBase::main() + 143
12  org.open-ephys.gui            	0x0000000102853143 juce::JUCEApplicationBase::main(int, char const**) + 83
13  libdyld.dylib                 	0x00007fff76ac93d5 start + 1

Thread 1 Crashed:: com.apple.audio.IOThread.client
0   org.open-ephys.gui            	0x00000001026bf6e4 SourceProcessorInfo::getSubProcessorIdx() const + 4
1   org.open-ephys.gui.plugin.LfpDisplayNode	0x0000000105160896 LfpViewer::LfpDisplayNode::process(juce::AudioBuffer<float>&) + 950
2   org.open-ephys.gui            	0x000000010279937d juce::GraphRenderingOps::AudioGraphRenderingOp<juce::GraphRenderingOps::ProcessBufferOp>::perform(juce::AudioBuffer<float>&, juce::OwnedArray<juce::MidiBuffer, juce::DummyCriticalSection> const&, int) + 141
3   org.open-ephys.gui            	0x00000001027854ba void juce::AudioProcessorGraph::processAudio<float>(juce::AudioBuffer<float>&, juce::MidiBuffer&) + 234
4   org.open-ephys.gui            	0x00000001027aa37a juce::AudioProcessorPlayer::audioDeviceIOCallback(float const**, int, float**, int, int) + 1418
[... 7 lines cut ...]
12  libsystem_pthread.dylib       	0x00007fff76cba249 _pthread_start + 66
13  libsystem_pthread.dylib       	0x00007fff76cb640d thread_start + 13

Thread 2:: Juce Timer
0   libsystem_kernel.dylib        	0x00007fff76c0186a __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff76cba56e _pthread_cond_wait + 722
2   org.open-ephys.gui            	0x00000001027f6b8e juce::WaitableEvent::wait(int) const + 190
3   org.open-ephys.gui            	0x0000000102858deb juce::Timer::TimerThread::run() + 235
4   org.open-ephys.gui            	0x00000001027f75b1 juce::Thread::threadEntryPoint() + 161
5   org.open-ephys.gui            	0x000000010281e65a threadEntryProc + 26
6   libsystem_pthread.dylib       	0x00007fff76cb72eb _pthread_body + 126
7   libsystem_pthread.dylib       	0x00007fff76cba249 _pthread_start + 66
8   libsystem_pthread.dylib       	0x00007fff76cb640d thread_start + 13

Thread 3:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff76bfe22a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff76bfe76c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff4a65113e __CFRunLoopServiceMachPort + 328
3   com.apple.CoreFoundation      	0x00007fff4a6506ac __CFRunLoopRun + 1612
4   com.apple.CoreFoundation      	0x00007fff4a64fe0e CFRunLoopRunSpecific + 455
5   com.apple.AppKit              	0x00007fff47cded1a _NSEventThread + 175
6   libsystem_pthread.dylib       	0x00007fff76cb72eb _pthread_body + 126
7   libsystem_pthread.dylib       	0x00007fff76cba249 _pthread_start + 66
8   libsystem_pthread.dylib       	0x00007fff76cb640d thread_start + 13
`

juulee2011, 2020-01-15:
Hello guys, I'm trying to use my own circuit to communicate with open-ephys, using serial port.
However, I've seen same problem as these articles, but there was no solution.
Is there anyone who solved this problem?
Thank you.

IFM-Ulm, 2020-01-15:
Hi,
we have not solved the issue but written a very very crude and rudimentary plugin which listens to a TCP-Port and is able to receive the data of an FPGA-board over TCP/IP and plot it. It does have some issues but it might point you into the right direction. It should be possible to use a virtual COM Port instead of TCP.
If you are interested, you can send  me a PM to joachim dot becker at uni minus ulm dot de
Cheers, Joachim

jsiegle, 2022-07-30:
The Serial Input plugin has been deprecated. If you need this plugin for your experiments, please open a new issue in the [`plugin-GUI`](https://github.com/open-ephys/plugin-GUI) repo.

closed 2022-07-30 by jsiegle