open-ephys/GUI #294: raw kwik file named  filename_100.raw.kwd

lepmik, 2015-06-22:
Is _100 meant to be?

aacuevas, 2015-06-22:
Yes, the GUI can record from different processors on the chain at once, creating one kwd file per processor, so the _number suffix indicates the processor ID the recording comes from.
We will eventually replace the ID numbers for more meaningful labels so it is easier to relate each file with its processor.

rossant, 2015-06-23:
I'm trying to open openephys files with [phy](https://github.com/kwikteam/phy/) (https://github.com/kwikteam/phy/issues/385)

currently it does not work because phy expects `/recordings/0/raw/hdf5_path` to be an attribute, not a group, with the path to an HDF5 file. There's no support to multiple HDF5 paths currently, because there can already be multiple recordings in one file.

anecdotally, we're also starting to move away from raw data in HDF5, because many users have their raw data in flat binary files, and that's just as easy to support. `/recordings/0/raw/` can contain a `dat_path` attribute which is a path to a `.dat` flat binary file. But in any case, phy will support both types of raw data.

aacuevas, 2015-07-15:
We do use multiple recordings, generating one every time recording is activated (either via the button or the record control node) but the GUI can also record from different points on the signal chain, hence the need for different files, each with multiple recordings.

I've updated the code so /recordigs/x/raw/hdf5_path is an attribute which points to the first file (usually the source node) and /recordigs/x/raw/hdf5_paths is a group that contains all the kwd files.

jsiegle, 2022-07-30:
The Kwik format has now been deprecated. We recommend switching to the [Binary Format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) instead.

closed 2022-07-30 by jsiegle