open-ephys/GUI #295: KWIK format incompatible with phy/Klusta-Suite latest format

nippoo, 2015-06-30:
Hi,

To quote @rossant:

> I'm trying to open openephys files with phy (kwikteam/phy#385)

> currently it does not work because phy expects /recordings/0/raw/hdf5_path to be an attribute, not a group, with the path to an HDF5 file. There's no support to multiple HDF5 paths currently, because there can already be multiple recordings in one file.

> anecdotally, we're also starting to move away from raw data in HDF5, because many users have their raw data in flat binary files, and that's just as easy to support. /recordings/0/raw/ can contain a dat_path attribute which is a path to a .dat flat binary file. But in any case, phy will support both types of raw data.

NB the latest version of the file format is http://phy.cortexlab.net/format/

open-ephys, 2015-07-02:
This should be easy to fix. Right now we're working on adding USB 3.0 support to the GUI, but we'll get to this as soon as that's ready.

aacuevas, 2015-07-15:
I've updated the hdf5_path issue, but, since we do not use yet many of the features available in a kwik file, we do not create some of the groups or attributes that might be needed for klustakwik to load the files properly. I'll update the code to create those empty elements until it loads.

jsiegle, 2022-07-30:
The Kwik format has now been deprecated. We recommend switching to the [Binary Format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) instead.

closed 2022-07-30 by jsiegle