open-ephys/GUI #298: Crash on hitting "record" and on returning from sleep, Mac version

macstry, 2015-07-03:
I've run into this issue with 0.3.5, but this has not occurred with 0.3.3. If I hit the record button without first hitting play, the program immediately crashes. I've also had the program crash on returning from sleep, I've posted an example of the latter crash message below.

----------------------------

Process:               open-ephys [4542]
Path:                  /Users/USER/Downloads/*/open-ephys.app/Contents/MacOS/open-ephys
Identifier:            org.open-ephys.gui
Version:               0.3.5 (7)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           open-ephys [4542]
User ID:               501

Date/Time:             2015-07-02 11:25:18.996 -0400
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        6BD2CA68-22AA-0623-D005-C745739A8D55

Time Awake Since Boot: 510000 seconds

Crashed Thread:        5  com.apple.audio.IOThread.client

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000004

VM Regions Near 0x4:
--> 
    __TEXT                 000000010b9dc000-000000010bd11000 [ 3284K] r-x/rwx SM=COW  /Users/USER/Downloads/*/open-ephys.app/Contents/MacOS/open-ephys

Thread 0:: Juce Message Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8a72b4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8a72a64f mach_msg + 55
2   com.apple.CoreGraphics        	0x00007fff93f8547c _CGSReenableUpdateForConnection + 93
3   com.apple.CoreGraphics        	0x00007fff93f85168 CGSUpdateManager::enable_update(unsigned long long) + 358
4   com.apple.AppKit              	0x00007fff97214029 -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 4951
5   com.apple.AppKit              	0x00007fff971cb2be -[NSView displayIfNeeded] + 1876
6   com.apple.AppKit              	0x00007fff971cab19 -[NSWindow displayIfNeeded] + 236
7   com.apple.AppKit              	0x00007fff971ca782 _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 936
8   com.apple.AppKit              	0x00007fff97900191 __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1540 + 46
9   com.apple.CoreFoundation      	0x00007fff8d94a127 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
10  com.apple.CoreFoundation      	0x00007fff8d94a080 __CFRunLoopDoObservers + 368
11  com.apple.CoreFoundation      	0x00007fff8d93c188 __CFRunLoopRun + 872
12  com.apple.CoreFoundation      	0x00007fff8d93bbd8 CFRunLoopRunSpecific + 296
13  com.apple.HIToolbox           	0x00007fff8dee056f RunCurrentEventLoopInMode + 235
14  com.apple.HIToolbox           	0x00007fff8dee02ea ReceiveNextEventCommon + 431
15  com.apple.HIToolbox           	0x00007fff8dee012b _BlockUntilNextEventMatchingListInModeWithFilter + 71
16  com.apple.AppKit              	0x00007fff971c79bb _DPSNextEvent + 978
17  com.apple.AppKit              	0x00007fff971c6f68 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
18  com.apple.AppKit              	0x00007fff971bcbf3 -[NSApplication run] + 594
19  org.open-ephys.gui            	0x000000010bb5300e juce::JUCEApplicationBase::main() + 190
20  org.open-ephys.gui            	0x000000010bb52f33 juce::JUCEApplicationBase::main(int, char const**) + 83
21  libdyld.dylib                 	0x00007fff900715c9 start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8a731232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff96cd3a6a _dispatch_mgr_thread + 52

Thread 2:: Juce Timer
0   libsystem_kernel.dylib        	0x00007fff8a730136 __psynch_cvwait + 10
1   org.open-ephys.gui            	0x000000010bb240ce juce::WaitableEvent::wait(int) const + 174
2   org.open-ephys.gui            	0x000000010bb5685f juce::Timer::TimerThread::run() + 463
3   org.open-ephys.gui            	0x000000010bb24b02 juce::Thread::threadEntryPoint() + 274
4   org.open-ephys.gui            	0x000000010bb38dea threadEntryProc + 26
5   libsystem_pthread.dylib       	0x00007fff906c9268 _pthread_body + 131
6   libsystem_pthread.dylib       	0x00007fff906c91e5 _pthread_start + 176
7   libsystem_pthread.dylib       	0x00007fff906c741d thread_start + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8a72b4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8a72a64f mach_msg + 55
[... 385 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         73.9M
__TEXT                            183.0M
__UNICODE                           552K
mapped file                        85.8M
shared memory                         4K
===========                      =======
TOTAL                               2.2G

Model: iMac14,3, BootROM IM143.0118.B09, 4 processors, Intel Core i7, 3.1 GHz, 16 GB, SMC 2.17f7
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.4f4 15601, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SD0128F, 121.33 GB
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: Expansion Desk
USB Device: Opal Kelly XEM6010-LX45
USB Device: USB20 Camera
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 23.10

chriswin2015, 2015-07-03:
> juce::OwnedArray const&, int) + 216
> +libhdf5_cpp.10.dylib (0) /opt/local/lib/libhdf5_cpp.10.dylib 0x10c238000 -
> 0x110da9000 - 0x110da9ff5 +cl_kernels (???) cl_kernels 0x110dab000 -
> 0x11abeb000 - 0x11ac41fff +libokFrontPanel.dylib (0) /Users/USER/Downloads/*
> 0x7fff8e270000 - 0x7fff8e274fff libpam.2.dylib (20) /usr/lib/libpam.2.dylib
> 0x7fff94769000 - 0x7fff94779ff7 libbsm.0.dylib (34) /usr/lib/libbsm.0.dylib
> 1.12) /System/Library/Frameworks/AudioUnit.framework/Versions/A/AudioUnit
> 675.42) /System/Library/Frameworks/QuickLook.framework/Versions/A/QuickLook
> 0x7fff9861a000 - 0x7fff98660ff7 libauto.dylib (186) /usr/lib/libauto.dylib

Hello. Your graphics interface coding has a syntax error search for the
following
" OpenCL.framework/Versions/A/OpenCL" and change the CL to GL. I read
through every line and i am very familiar with Juce and Open-Ephys. I know
how to use it with robotic control modules with AI. I will double check
everything though. I may find another error.

chriswin2015, 2015-07-03:
I read through everything again. It is in fact your gui module that has an
IO problem which is directly related to how your OpenGL is being compiled.
If that does not work i would download and install the open-ephys gui
module. This should also fix your problem. Whoever made this script does
not organize very well. I will do some further digging if these two things
do not work

macstry, 2015-07-07:
I downloaded and installed the current open-ephys gui module (0.3.5), opened it, dragged the FPGA to the signal chain along with an LFP sink and hit record with an immediate crash. Perhaps this is the same error that was occurring in the Linux version? Here's the crash report:
------------
Process:               open-ephys [8791]
Path:                  /Users/USER/Downloads/*/open-ephys.app/Contents/MacOS/open-ephys
Identifier:            org.open-ephys.gui
Version:               0.3.5 (0.3.5)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           open-ephys [8791]
User ID:               501

Date/Time:             2015-07-07 11:27:41.160 -0400
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        6BD2CA68-22AA-0623-D005-C745739A8D55

Time Awake Since Boot: 940000 seconds

Crashed Thread:        8  com.apple.audio.IOThread.client

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT

Thread 0:: Juce Message Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8a730166 __psynch_mutexwait + 10
1   org.open-ephys.gui            	0x00000001059819d9 0x105845000 + 1296857
2   org.open-ephys.gui            	0x000000010594a3e3 0x105845000 + 1070051
3   org.open-ephys.gui            	0x000000010599e63a 0x105845000 + 1414714
4   com.apple.CoreFoundation      	0x00007fff8d94aa01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
5   com.apple.CoreFoundation      	0x00007fff8d93cb8d __CFRunLoopDoSources0 + 269
6   com.apple.CoreFoundation      	0x00007fff8d93c1bf __CFRunLoopRun + 927
7   com.apple.CoreFoundation      	0x00007fff8d93bbd8 CFRunLoopRunSpecific + 296
8   com.apple.HIToolbox           	0x00007fff8dee056f RunCurrentEventLoopInMode + 235
9   com.apple.HIToolbox           	0x00007fff8dee01ee ReceiveNextEventCommon + 179
10  com.apple.HIToolbox           	0x00007fff8dee012b _BlockUntilNextEventMatchingListInModeWithFilter + 71
11  com.apple.AppKit              	0x00007fff971c79bb _DPSNextEvent + 978
12  com.apple.AppKit              	0x00007fff971c6f68 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
13  com.apple.AppKit              	0x00007fff971bcbf3 -[NSApplication run] + 594
14  org.open-ephys.gui            	0x000000010599b41e 0x105845000 + 1401886
15  org.open-ephys.gui            	0x000000010599b343 0x105845000 + 1401667
16  libdyld.dylib                 	0x00007fff900715c9 start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8a731232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff96cd3a6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8a72b4de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8a72a64f mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff8d93ceb4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff8d93c37b __CFRunLoopRun + 1371
4   com.apple.CoreFoundation      	0x00007fff8d93bbd8 CFRunLoopRunSpecific + 296
5   com.apple.AppKit              	0x00007fff9728f66b _NSEventThread + 137
6   libsystem_pthread.dylib       	0x00007fff906c9268 _pthread_body + 131
7   libsystem_pthread.dylib       	0x00007fff906c91e5 _pthread_start + 176
8   libsystem_pthread.dylib       	0x00007fff906c741d thread_start + 13

Thread 3:: Juce Timer
0   libsystem_kernel.dylib        	0x00007fff8a730136 __psynch_cvwait + 10
1   org.open-ephys.gui            	0x000000010596ed0e 0x105845000 + 1219854
2   org.open-ephys.gui            	0x000000010599ee1f 0x105845000 + 1416735
3   org.open-ephys.gui            	0x000000010596f762 0x105845000 + 1222498
4   org.open-ephys.gui            	0x00000001059825aa 0x105845000 + 1299882
5   libsystem_pthread.dylib       	0x00007fff906c9268 _pthread_body + 131
6   libsystem_pthread.dylib       	0x00007fff906c91e5 _pthread_start + 176
7   libsystem_pthread.dylib       	0x00007fff906c741d thread_start + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8a73094a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff906c740d start_wqthread + 13
[... 344 lines cut ...]
__TEXT                            161.6M
__UNICODE                           552K
mapped file                        68.2M
shared memory                         4K
===========                      =======
TOTAL                             640.5M
TOTAL, minus reserved VM space    640.0M
 

Model: iMac14,3, BootROM IM143.0118.B09, 4 processors, Intel Core i7, 3.1 GHz, 16 GB, SMC 2.17f7
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.4f4 15601, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SD0128F, 121.33 GB
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: Expansion Desk
USB Device: Opal Kelly XEM6010-LX45
USB Device: USB20 Camera
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 23.10

jsiegle, 2022-07-30:
Just tested this on Mac, it doesn't seem to be a problem any more.

closed 2022-07-30 by jsiegle