open-ephys/GUI #299: Segfault when attempting to record with ADC

simonster, 2015-07-03:
This is my first time using the OpenEphys system, so it's possible I'm not doing something right, but the first thing I tried is segfaulting. Here's what I did:

1. Launch OpenEphys (compiled from source on Ubuntu 14.04, commit a52dbdb229be8157c7690c17e17dd5abbf4462e0)
2. Drag Rhythm FPGA to the bottom
3. Click ADC 1-8 button (I don't have any headstages connected, and there's actually nothing plugged into the ADC ports, if that makes a difference)
4. Click record button

Here's the output from gdb with the backtrace: https://gist.github.com/simonster/7aeeb7c8b4777cc96c29

open-ephys, 2015-07-03:
It looks like this is a real problem. The cause of it is hitting "record" immediately, rather than acquisition first, then record. Can you confirm that there's no segfault if you acquire prior to recording?

This should be easy enough to fix...hitting record right off the bat has definitely worked in the past.

simonster, 2015-07-03:
If I start acquisition first there is no segfault. That works for now. (As I said, I don't know what I'm doing yet.) Thanks for looking into this!

chriswin2015, 2015-07-03:
What are you using as a serial bus?

simonster, 2015-07-03:
@chriswin2015 It's just connected via USB, if that's what you mean.

aacuevas, 2015-07-03:
I have located where the issue is. It's caused by the synchronization messages we store when recording starts not having a valid timestamp until acquisition has started. I'll probably be able to have it fixed this weekend.

aacuevas, 2015-07-15:
It should be fixed now, please check that it works for you too.

closed 2022-07-30 by jsiegle