open-ephys/GUI #304: Add checks for 100% cpu load and aq. board buffer overruns

jvoigts, 2015-07-23:
100% cpu load spikes can lead to intermittent, and non-obvious data loss.

Add two maybe three checks to the GUI and trigger warnings (and add them to the logs) automatically. This is related to some users reporting data corruption following 100% cpu load spikes. (see here  https://groups.google.com/d/topic/open-ephys/JKG0xTxO1Ig/discussion)

- Warning on 100% CPU load:  temporarily stalling the system will cause issues in the USB pipeline, this could possibly lead to file integrity issues. GUI should warn user and add note to log.

- Aq. board buffer: this is a related issue because the buffer would only fill up at all if there's a temporary break in the data polling by the host PC, but we can check the state of the buffer in the GUI and use this as a clean way to detect if this problem occurred.

- Possibly check if sample count/timestamps are saves without gaps - this check could be tricky to implement, and the other two should catch most cases of data loss, so maybe we can push this check to the data import function in matlab/python/whatever and just warn the user (again) when they load something that has gaps - also the ts interpolation should be able to deal with this gracefully.

jvoigts, 2015-07-23:
Related issue #143 ( High CPU usage by other software can cause crashes) should be checked after this, ideally even 100% load should never crash, but only result in temporary data loss that is appropriately indicated to the user, and noted in the log.

jsiegle, 2022-07-30:
The Record Node now includes an internal buffer to prevent data loss, even if CPU usage spikes. If this buffer becomes more than 90% full, recording will automatically stop.

closed 2022-07-30 by jsiegle