open-ephys/GUI #306: Outdated GitHub wiki should warn readers on more than its front page

galenlynch, 2015-08-01:
Many google searches about Open Ephys list this GitHub wiki before the actively maintained Atlassian wiki, and bypass the GitHub wiki's front page. This can (and in my case did) lead to situations where users searching for information find the outdated wiki first, and are unaware that it is outdated. For example, when I tried to find more information about the Open Ephys data format, it took me a while to realize that what I was reading and basing code off of was not longer current.

I don't know to what extent this is possible, but it would be nice to have every wiki page warn readers that a more current wiki exists, and that the GitHub wiki is no longer being updated.

open-ephys, 2015-08-03:
We can probably just delete the old wiki at this point. Any objections?

open-ephys, 2015-08-05:
Ok, the original wiki is gone!

closed 2015-08-05 by open-ephys