open-ephys/GUI #307: ADC1-7 almost identical

benglitz, 2015-08-18:
Dear community,

I am encountering a problem when recording both neural data via the headstage (intan 64) and ADC channels. Both neural data and ADC data is correctly displayed. The neural data is also saved correctly, but the SAVED ADC channels 2-7 are almost identical to ADC 1. ADC1 has a trigger connected to control the recording, and ADC2 another one, both via the HDMI connected board. The information from the latter (ADC2) is lost. This is running the latest version of the code, compiled on Windows 7.

If anyone has seen a solution to this please let me know.

Best wishes,

B

closed 2015-08-18 by benglitz

open-ephys, 2015-08-18:
Did you figure out what the problem was?

benglitz, 2015-08-18:
Yes. The problem was on my end, since only the Digital In but not the ADC channels were connected at that moment. It may make sense to delete this issue, to avoid confusion.