open-ephys/GUI #310: Assertion failure on TTL input from Pulse Pal

lepmik, 2015-08-31:
Hi,

I get a crash when sending a a burst of pulses to the TTL input but not when I stream the pulses to the Analog input.

```Bash
Adding audio callback. # Starting acquisition is fine, but after some tens of pulses I get
JUCE Assertion failure in juce_AudioSampleBuffer.cpp:531 # The LFP display freezes but GUI works

Removing audio callback. # After pressing the play button to stop acquisition the GUI freezes totally
```

open-ephys, 2015-08-31:
What's the frequency of the pulses?

lepmik, 2015-09-01:
It's a 20 Hz pulse train on/off for one sec

open-ephys, 2015-09-01:
Does it happen every time you send the pulses, or just occasionally? Also, which processors are in your signal chain?

lepmik, 2015-09-02:
It happens every time, mostly after the first pulse train. 
Signal chain:
A: Rhythm FPGA > LFP Viewer with only the ADC's on
B: Serial Port > Pulse Pal

open-ephys, 2015-09-02:
Can you try it with a headstage connected? There could be an issue with assigning timestamps to the pulses (which are converted to TTL events as soon as they arrive) when there's no headstage data. This is definitely something we can fix, using a headstage might solve the problem for now.

lepmik, 2015-10-02:
Sorry for the delay. Connecting a headstage does make it more stable,
however I still get the same crash after sending several trains.

lepmik, 2015-10-26:
Hi, have you been able to look into this yet?

jsiegle, 2022-07-30:
The Pulse Pal plugin source code now lives here: https://github.com/open-ephys-plugins/pulse-pal-output

If you're still experiencing any problems with this plugin, please open a new issue there.

closed 2022-07-30 by jsiegle