open-ephys/GUI #312: How to send strings to the Network Events module via TCP using Matlab 2007b

km2015, 2015-09-09:
Hi there,

I am trying to send strings to the Network Events module via TCP using Matlab 2007b. My first question is : Is it possible ?

When I run the compile_matlab_wrapper.m file, the first step (e.g. 1. make sure mex is setup properly and a compiler is available)  works but the second step doesn't (compiling step) doesn't.

Matlab displays the following error message (see below). Did someone have the same problem ?

Thx

Karin

____

This is mex, Copyright 1984-2006 The MathWorks, Inc. 
 
Warning: libzmq-v110-mt-3_2_2.dll specified with -l option not found on -L path
Default options filename found in C:\Users\huberlab\AppData\Roaming\MathWorks\MATLAB\R2007b 
---------------------------------------------------------------- 
->    Options file           = C:\Users\huberlab\AppData\Roaming\MathWorks\MATLAB\R2007b\mexopts.bat 
      MATLAB                 = C:\PROGRA~2\MATLAB\R2007b 
->    COMPILER               = lcc 
->    Compiler flags: 
         COMPFLAGS           = -c -Zp8 -I"C:\PROGRA~2\MATLAB\R2007b\sys\lcc\include" -DMATLAB_MEX_FILE -noregistrylookup 
         OPTIMFLAGS          = -DNDEBUG 
         DEBUGFLAGS          = -g4 
         arguments           =  -Ic:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include 
         Name switch         = -Fo 
->    Pre-linking commands   =  
->    LINKER                 = lcclnk 
->    Link directives: 
         LINKFLAGS           =  -tmpdir "." -dll "C:\PROGRA~2\MATLAB\R2007b\extern\lib\win32\lcc\mexFunction.def" -L"C:\PROGRA~2\MATLAB\R2007b\sys\lcc\lib" -libpath "C:\PROGRA~2\MATLAB\R2007b\extern\lib\win32\lcc" C:\Users\huberlab\AppData\Local\Temp\mex_2548A765-4BC1-4322-23A3-486A3CCADF0E\templib2.obj 
         LINKDEBUGFLAGS      =  
         LINKFLAGSPOST       = libmx.lib libmex.lib libmat.lib 
         Name directive      = -o "zeroMQwrapper.mexw32" 
         File link directive =  
         Lib. link directive =  
         Rsp file indicator  = @ 
->    Resource Compiler      = lrc -I"C:\PROGRA~2\MATLAB\R2007b\sys\lcc\include" -noregistrylookup -fo"mexversion.res" 
->    Resource Linker        =  
---------------------------------------------------------------- 
 
 
--> "lcc  -Ic:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include -c -Zp8 -I"C:\PROGRA~2\MATLAB\R2007b\sys\lcc\include" -DMATLAB_MEX_FILE -noregistrylookup -FoC:\Users\huberlab\AppData\Local\Temp\mex_2548A765-4BC1-4322-23A3-486A3CCADF0E\zeroMQwrapper.obj -IC:\PROGRA~2\MATLAB\R2007b\extern\include -DNDEBUG -DMX_COMPAT_32 C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp" 
 
lcc preprocessor error: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h:87 C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp:9 Could not find include file <stdint.h> 
lcc preprocessor error: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq_utils.h:38 C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp:10 Could not find include file <stdint.h> 
lcc preprocessor error: C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp:13 Could not find include file <string> 
lcc preprocessor error: C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp:14 Could not find include file <queue> 
lcc preprocessor error: C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp:15 Could not find include file <list> 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 161  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 167  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 170  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 185  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 186  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 187  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 188  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 189  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 192  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 193  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 194  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 205  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 206  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 207  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 209  unrecognized declaration 
Error C:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp: c:\DATA\GitHub\GUI\Resources\windows-libs\ZeroMQ\include\zmq.h: 210  too many errors 
 
  C:\PROGRA~2\MATLAB\R2007B\BIN\MEX.PL: Error: Compile of 'c:\DATA\GitHub\GUI\Resources\Matlab\windows\zeroMQwrapper.cpp' failed. 
 
??? Error using ==> mex at 208
Unable to complete successfully.

m4loman, 2015-10-01:
there is some wierdness with the original script.

try to do something like this and see if it works. make sure both zeroMQwrapper.cpp and libzmq-v120xxx are in the dir where you are running this.

mex 'zeroMQwrapper.cpp' -I'XXX\GUI-master\Resources\windows-libs\ZeroMQ\include' -L'XXX\GUI-master\Resources\windows-libs\ZeroMQ\lib_x86\' -l'libzmq-v120-mt-4_0_4'

balajisriram, 2015-12-11:
Hi,
Was this resolved? what was the solution? Any chance I can get a copy of zeroMQWrapper.mexw32 if its available?

m4loman, 2015-12-11:
No problem:
I will upload it in the same location where the 64bit version is.

FYI in case you want to compile yourself:
here is an example of what the command would look like

mex 'zeroMQwrapper.cpp' -I'C:\GUI-master\Resources\windows-libs\ZeroMQ\include' -L'C:\\GUI-master\Resources\windows-libs\ZeroMQ\lib_x86\' -l'libzmq-v120-mt-4_0_4'

m4loman, 2015-12-11:
e voila

balajisriram, 2015-12-12:
Which compiler did you use?

m4loman, 2015-12-12:
Visual studio express 2013

km2015, 2015-12-12:
hi there,
I put screenshots of my notes. Do you have to use Matlab 2007 also ?

[image]
[image]
[image]

jsiegle, 2022-07-30:
The Network Events plugin now only supports sending TTL events (up to 256 lines).

To send text-based messages to all plugins, you can use the `/api/message` endpoint of the Open Ephys HTTP Server.

closed 2022-07-30 by jsiegle