open-ephys/GUI #315: improve user friendliness of audio settings menu

jvoigts, 2015-10-15:
Right now, its not clear how to get to the setting menu (i think you still have to click the small X ms number). This should be a button and also be accessible form the dropdown (file or edit?) menu.

Also, the audio buffer size setting is a pretty important setting that affects all closed-loop applications and can affect the function of plugins if they're not written to be buffer-size independent. This should be clearer to users - probably just re-naming the fields would help a lot.

open-ephys, 2015-10-15:
This won't be trivial because the Audio Settings menu is a standard JUCE class. But I agree that this is an important thing to change.

aacuevas, 2015-10-15:
I was just thinking about this, and I believe that at some point, it might be interesting to decouple the signal thread from the audio system. It's far from trivial but it would grant us more control over the buffers and latencies. As well as allowing the GUI to work on systems that might not have an audio card properly configured.

jvoigts, 2015-10-15:
Ah yea i remember the juce class issue.. Can we maybe call a JUCE method to set the buffer size and make at least a secondary menu?

aacuevas, 2015-10-15:
We could always code our own dialog for that instead of using the one Juce provides, it's not mandatory, just a convenience. It's a bit of work but nothing that can't be done.

closed 2022-07-30 by jsiegle