open-ephys/GUI #316: LfpDisplayNode crashes, might be fixed

jvoigts, 2015-11-05:
JUCE Assertion failure in juce_AudioSampleBuffer.cpp:531
On Mac OS 10.10 and use the code from main branch in github
This seems to be caused by LfpDisplayNode, where the bufferIndex variable in handleEvents can get negative under certain conditions. 
I have posted a pull request with a possible fix, based on my understanding of the logic of that method. 
With the fix we are not experiencing the error any more. 

see also https://groups.google.com/forum/#!topic/open-ephys/7Jdkkhjj9u4

malfatti, 2015-11-22:
Hello!

I have the same trouble, could you send me your fixed script/guide me through your solution? I'm trying to fix this for an experiment we will run soon.

Thanks a lot!

jvoigts, 2015-11-22:
have you tried recompiling after the Aug.30 fix that francesco posted? This 5be77101acbf30679a558a70a4c7699236e2f44f is the fix.

malfatti, 2015-11-23:
No, I haven't! Now I tried and it is working perfectly. Thanks a lot for pointing it out!

I was using the rha2000-eval branch, maybe I should submit this fix to that branch too.

Cheers!

jvoigts, 2015-11-23:
sweet, thx @fpbattaglia !

closed 2015-11-23 by jvoigts