open-ephys/GUI #320: USB3 impedance/spike sorter error

nikolaskaralis, 2015-11-19:
Using the USB3 FPGA on Linux, I can record normally.
However, trying to measure impedance, I get in the console: 
"Error in RHD2000EvalBoard"
for each channel, and then some crazy values as impedance (on the GUI).
Also trying to use Spike Sorter (with usb3) crashes the GUI.

On a perhaps related note: While recording with USB3 FPGA on Linux works fine (except of the above), on Windows (on the same computer, dual boot) the headstages are not even recognized.
USB2 board works for both systems.
 
Is anybody working with the USB3 board?

Cheers,
Nikolas

aacuevas, 2015-11-19:
-Does it say just "Error in RHD2000EvalBoard"? usually those messages are followed by "::somemessage" which is important to locate where it is failing.
-Are you using the precompiled or the source version? Can you try with the other?
-How many channels are you acquiring? The spike sorter code is completely independent of the source, but there might be an issue with high channel counts.

The windows issue is quite weird. Is it not recognizing just the headstages, or the FPGA as a whole? Can you paste the console log? If it recognized the FPGA but not the headstages, try changing in rhd2000evalboard.h line 33 the 2048 value for 1024 and see if it solves the issue.

Aarón.

nikolaskaralis, 2015-11-21:
As you can see in the attached logs, the error is: 
"Error in Rhd2000EvalBoard::readDataBlock: Incorrect header."
I tried with both precompiled and source version and it gives the same problem.
Also tried in both Ubuntu and Windows10 and has the same problem with the impedance measurement.
I am using 16/32 channels (2 headstages).
I could not replicate the issue with the spike sorter this time (the logs and problem is with impedance measurement).

As for the windows issue, I still have this issue on that laptop (Windows 7) but tried with another laptop (Windows10) and it worked so not sure what is the problem there. I will try with another Windows7 computer just to make sure it is not a version issue.

[open_epys_impedance_log_ubuntu.txt](https://github.com/open-ephys/GUI/files/40804/open_epys_impedance_log_ubuntu.txt)
[open_epys_impedance_log_windows10.txt](https://github.com/open-ephys/GUI/files/40805/open_epys_impedance_log_windows10.txt)
[image]

aacuevas, 2015-11-24:
I've fixed the impedance measurement, should work now.
I've also reduced the USB3 block size to 1024, as the higher 2048 value causes problems with some drivers. Please check if this solves your problem on Windows.

shayo, 2015-11-24:
Aaron, I have a couple of questions regarding the new USB3 feature:
1) I saw a new .bit file. I assume there is new verilog code? Where is that stored in the repository?
2) You said that existing break out boards can still be used, and only the FPGA needs to be replaced, is that correct?
3) How can one connect four 32ch amplifier to one port? Is there existing PCB/cabling solution for this?

Thanks!

Shay

aacuevas, 2015-11-24:
Hi Shay,
In order:
1) The usb3 code is stored in the "usb3" branch of our Rhythm repo ( https://github.com/open-ephys/rhythm )
2) That is correct, just replacing the XEM6010-LX45 (usb2) for the XEM6310-LX45 (usb3) is enough.
3) You can't connect four 32ch headstages to one port, sadly. You can, however, connect either two 64ch headstages using Intan's dual headstage adapter or one 128ch headstage.

jvoigts, 2015-11-24:
1) I saw a new .bit file. I assume there is new verilog code? Where is that stored in the repository?

should be https://github.com/open-ephys/rhythm/tree/usb3

Aaron had to make a lot of fairly deep changes, i dont know if there's a list/documentation of this somewhere already. If not it might be useful to do that at some point, similar to https://open-ephys.atlassian.net/wiki/display/OEW/Rhythm+firmware+fork

2) You said that existing break out boards can still be used, and only the FPGA needs to be replaced, is that correct?

yea, the I/O boards are the same you just swap out the usb3 fpga board

3) How can one connect four 32ch amplifier to one port? Is there existing PCB/cabling solution for this?

You can only connect two RHD headstages per port, using a Y-adapter, because there's only 2 miso pairs per connector. This gives you 8 RHD chips total, for up to 512 channels (with 64 ch chips). 

We have a few free pins left on the fpga that we pulled out to a .1" header for future expansions so with a few cmos<>lvds bridges (because we dont have enough pins for straight lvds - and this is assuming that the bandwidth of these is not destroyed by the board layout etc) we can theoretically throw in a 2nd row of 4 ports, which would get us to 1024 chs per board.  Of course this would require some firmware, api, and GUI updates, and i'm not 100% sure how close this would come to saturating the usb3, though it should fit in theory ( http://www.wolframalpha.com/input/?i=2+byte+*+30kHz+*+1024+*+2+in+MB%2Fs ).

I started putting together an extension board in the https://github.com/open-ephys/acquisition-board repository, if someone is up for some firmware coding this could be a fun project.

shayo, 2015-11-24:
Got it. Thanks for the info!

nikolaskaralis, 2015-11-26:
Hi Aaron,
the impedance problem is fixed now.
I will check about the Windows issue and update here.

closed 2022-07-30 by jsiegle