open-ephys/GUI #322: Arduino Output initialization issue

jperge, 2015-11-23:
Hi,

I'm having trouble initializing the Arduino Output module (following the Phase Detector module in the signal chain). The error message I get is 'Cannot change device while acquisition is active'. 

I found a way to work around this by starting and stopping the data acquisition, set the Arduino Output port to 'Device' then back to the arduino port (COM4 in my setup). After this Arduino does initialize. 

This is a little inconvenient to change every time and the bug could be fixed by initializing the module properly. Please let me know how to fix this issue in the code. 

Thanks,

Janos

aacuevas, 2015-11-24:
There was an uninitialized variable that was causing this. I've just uploaded a patch, please check if it correctly fixes the issue for you.

jperge, 2015-11-24:
Yes it did! Problem solved! 

Thank you!

closed 2015-11-24 by jperge