open-ephys/GUI #325: std::endl overuse

jonnew, 2015-12-18:
In general, `<< std::endl` should not be used to introduce a line ending into to an output stream because it is syntactic sugar for `<< '\n' << std::flush`, which clears the buffer and introduces unnecessary overhead. In almost all circumstances, `\n` is fine. The compiler will configure exact characters to indicate a new line depending on the OS.

I'm submitting this issue because I see tons of  `<< std::endl` in the portions of the code I'm working with.

jsiegle, 2015-12-18:
That's what I used almost everywhere. Any reason not to do a find & replace for std::endl to '\n'?

aacuevas, 2015-12-18:
While that's true, most of the debug messages we send to the console are done in non-critical initialization rutines, so the performance overhead is negligible, except perhaps in a couple of specific places (I'm thinking the open file messages when recording starts).
Also, on some systems without manually flushing the buffer doesn't get printed to the screen until it fills or the OS decides it's time to print it, which might take a while. If we were to replace all endl we should manually add some flushes at critical points to be sure that the cummulative debug lines are printed at certain points.
That's specially important for debugging, as any missing line still in the buffer when users report their console contents might difficult bug solving.

Slightly related, long term we should add some kind of --verbose command line argument and simply avoid printing most , if not all, of those lines unless it's specified.

jonnew, 2015-12-19:
OK, I think a resolution is to simply remove gratuitous cases when we see them:
```
std::cout << "foo" << std::endl;
std::cout << "bar" << std::endl;
``

With regard to the `--verbose` option: yes, that is good idea.

jsiegle, 2022-07-30:
The GUI now includes a built-in logger for more fine-tuned control of debugging output. See the [Open Ephys Plugin API documentation](https://open-ephys.github.io/gui-docs/Developer-Guide/Open-Ephys-Plugin-API/index.html#logging) for details.

closed 2022-07-30 by jsiegle