open-ephys/GUI #327: Raw timestamps in Kwik format

mmyros, 2015-12-28:
This was a question in the forum that never received an answer, I am wondering about this now. Raw timestamps are very useful in the .continuous files, but are they in .kwik?

"...where can I find the rawTimestamps on an experiment saved in kwik format? They don't seem to be anywhere in the kwd file."

aacuevas, 2015-12-28:
The kwik format specification does not include raw timestamps, just the start sample and it assumes that all the samples in a recording are continuous (which is usually true, unless there is some buffering issues).

Adding those as a custom extension to the format is something I plan to do, along some other additions to the saving formats (like saving the whole TTL word)

jsiegle, 2022-07-30:
The Kwik format has now been deprecated. We recommending using the [Binary Format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) instead.

closed 2022-07-30 by jsiegle