open-ephys/GUI #328: Crash after stop recording in Kwik format

hstemmann, 2016-01-07:
Hi,

the GUI crashes as soon as I stop the recording. This only happens when storing data in Kwik format and only when I have a filter node in my signal chain (i.e. Bandpass Filter, Spike Detector, Spike Sorter, Channel Map, Common Avg Ref, apparently Phase Detector does not seem to be a Problem).
Storing the data in the OpenEphys format does not show this behavior. Is there anything that can be done to overcome this problem?
Thanks so much for your help.

Here are the first few lines of the detailed problem Report (I could as well provide the full report):

Process:               open-ephys [67472]
Path:                  /Users/USER/Documents/*/open-ephys.app/Contents/MacOS/open-ephys
Identifier:            org.open-ephys.gui
Version:               0.3.4 (0.3.4)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           open-ephys [67472]
User ID:               501

Date/Time:             2016-01-07 14:33:52.555 +0100
OS Version:            Mac OS X 10.11.2 (15C50)
Report Version:        11
Anonymous UUID:        09C7F220-58CE-EF2A-22D5-BEF4F2C13D0D

Sleep/Wake UUID:       BCE506FD-90E2-409E-99E2-AAFB6242C039

Time Awake Since Boot: 190000 seconds
Time Since Wake:       12000 seconds

System Integrity Protection: enabled

Crashed Thread:        14  com.apple.audio.IOThread.client

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x28:
--> 
    __TEXT                 0000000108ab6000-0000000109409000 [ 9548K] r-x/rwx SM=COW  /Users/USER/Documents/*/open-ephys.app/Contents/MacOS/open-ephys

aacuevas, 2016-01-08:
Hi,

Can you please open the software via the console and paste the few last lines when a crash occurs? Also, could you please save a signal chain that causes the crash and attach the saved chain file?

Best,
Aarón.

hstemmann, 2016-01-08:
Hi Aarón,

thanks for your response. Here are the last few line after a crash happened and attached you will find the signal chain.

Best,

Heiko

_________

[Crash_BandPassFilter.zip](https://github.com/open-ephys/GUI/files/82612/Crash_BandPassFilter.zip)

Signal chain: 0

Source node: Rhythm FPGA.
     Connecting to audio and record nodes.
     Connecting Rhythm FPGA 100 to Bandpass Filter 102

Source node: Bandpass Filter.
     Connecting to audio and record nodes.
     Connecting Bandpass Filter 102 to LFP Viewer 101

Source node: LFP Viewer.
     NOT connecting to audio and record nodes.
     No dest node.

Enabling processors...
Source node received enable signal
Expecting 35 channels.
Number of 16-bit words in FIFO: 0
Is eval board running: 0
Starting acquisition.
Flushing FIFO.
Resizing buffer. Samples: 150000, Inputs: 35
   Enabling VisualizerEditor
Beginning animation.
Filter Viewport disabled.

Adding audio callback.
Creating new directory.
Sources/Rhythm FPGA
Create subnodes with parameters
  Moving forward along signal chain.
Filters/Bandpass Filter
Create subnodes with parameters
  Moving forward along signal chain.
Sinks/LFP Viewer
Create subnodes with parameters
  Moving forward along signal chain.
  No processor found.
  End of chain.
Segmentation fault: 11

______

aacuevas, 2016-02-03:
Hi,

Sorry for the long silence.

I can't reproduce the crash. What are the steps? Just start recording and then stop or there is any other step that causes the crash?

However, the save file you sent me is for a much older version of the GUI (0.3.4). Have you tried with the latest version (0.3.6) to see if the issue is still there? Much code has been changed since then.

hstemmann, 2016-02-08:
Hi,

I have tried the latest version now - the problem does not occur anymore. Thanks for your support.

aacuevas, 2016-02-08:
Glad to see the issue is solved. I'm clossing it then.

closed 2016-02-08 by aacuevas