open-ephys/GUI #329: Crash if the recording folder is not found

nikolaskaralis, 2016-02-03:
When trying to record in a folder (saved and loaded from the last recording settings) and the folder is not found the GUI crashes.
In my particular case the folder is on another partition and after computer restart it seems that although it is mounted, unless you explicitly open a folder inside the partition the GUI doesn't recognize it or is not able to write there.

So maybe it would be good if there was a check if path exists and is writable and if not, give an error but not crash.

Tested in Ubuntu

aacuevas, 2016-02-08:
Yup, we just assume the path is writable (I thought Juce also took care of creating the needed parent folders, but perhaps it doesn't in every case). It's easy to just add a couple of checks and revert to the default path, with a warning, if the one set is not available or writable.

kushagra06, 2016-03-20:
@nikolaskaralis @aacuevas : I am able to record in a saved and loaded folder without any crash. Can you provide me more details?

closed 2022-07-30 by jsiegle